dartsim/dart #1203: libode-dev is required for build from source

eastskykang, 2018-12-10:
I'm building dart release 6.6.2 version from source and it seems dart requires libode-dev library but it's omitted from installation manual: https://dartsim.github.io/install_dart_on_ubuntu.html

Is it deprecated in the latest version?

mxgrey, 2018-12-11:
If I'm not mistaken, I think `libode-dev` should only be necessary for the optional library component `libdart-collision-ode`. The parts of ODE that DART has a hard dependency on should come with the source code in the folder `dart/external/odelcpsolver`.

Could you put your configuration and/or build errors in a [gist](https://gist.github.com/) and link them here?

eastskykang, 2018-12-12:
@mxgrey  thank you for your response! 

Yes, I know that is optional. I just wanted to suggest add libode-dev to install guide, because it might be confusing to users.

jslee02, 2018-12-16:
@eastskykang Thanks for the report. I added `libode-dev` to the installation instruction.

@mxgrey The installation guide was outdated, but it's now updated. :smile:

closed 2018-12-16 by jslee02