dartsim/dart #1204: Real-time window simulation

adalessandro, 2018-12-13:
Running the tutorial *tutorialBiped*, the number of frames per second I get is ~500, causing a *slow* (not real-time) simulation. Is it supposed or is there a way to run the simulation in real-time?

AFAICS, the way SimWindow class drives the simulation, it is not guaranteed to run in real-time.
The SimWindow class computes the number of iterations (steps) based on the value of `mDisplayTimeout`.
https://github.com/dartsim/dart/blob/9a6bb46896ddc1b073e1b0d73f4d69d4fe321f1f/dart/gui/glut/SimWindow.cpp#L128

The next callback is registered using `glutTimerFunc`, which will be triggered in **at** **least** `mDisplayTimeout` milliseconds, but could be more than that.
https://github.com/dartsim/dart/blob/9a6bb46896ddc1b073e1b0d73f4d69d4fe321f1f/dart/gui/glut/SimWindow.cpp#L140

Shouldn't SimWindow compute the number of steps based on the real time elapsed from the last update callback?

This might be related to https://github.com/dartsim/dart/issues/735

mxgrey, 2018-12-14:
> This might be related to #735

Issue #735 is an integer->double conversion bug that halts simulation altogether due to rounding down to `0`. It wouldn't make simulation slower; it just stops simulation altogether.

The GUI doesn't make any effort to match real time during a simulation run. The GUI is just there to show you what's happening in the simulation at whatever speed the simulation happens to be running. Simulation speed can depend on many factors, like the size of the time step that you've chosen (the default being `0.001s`) or the complexity of your simulation (e.g. how many collidable objects are there; how many joints are being simulated).

Since you have the ability to program your GUI application, you can choose to program the GUI to play back at real time. If your simulation runs faster than real time, then you can have your GUI throttle the simulation. If your simulation runs slower than real time, then you can either:
1. Run the full simulation, record the states, and replay those states at real time
2. Tune your simulation so that it can run faster than real time, and then throttle it

These aren't features that dartsim offers out of the box.

mxgrey, 2018-12-14:
Also, if you want to make a sophisticated GUI application with dartsim, I would recommend using the `dart::gui::osg` component library which uses [OpenSceneGraph](http://www.openscenegraph.org/). It's much more user-friendly than the glut component library.

adalessandro, 2018-12-14:
@mxgrey , thanks for the explanation. It is clear to me now that this is the intended behavior for the tutorials.

However, IMHO the expected behavior one might expect to see in a GUI example by default is a real-time simulation loop. It's just a comment, we can close this.

closed 2018-12-19 by adalessandro

mxgrey, 2019-01-11:
Just thought I'd note: A real-time simulation display has been added in release 6.7.0: #1216