dartsim/dart #1210: setacceleration not working

mamurtaza88, 2019-01-08:
### Bug Report
> Please answer the following questions for yourself before submitting an issue.
- [ yes] I checked the [documentation](http://dartsim.github.io/) and found no answer.
- [ Yes] I checked to make sure that this issue has not already been filed.

#### Environment
* Select DART version: (6.6.2)
* Select OS name and version name(or number): Ubuntu 16.02
* Select compiler name and version number: GCC 5.4.0 20160609

#### Expected Behavior
> We are trying to pass accelerations to joint to KR5 robot. We expect that joints should be moving with that accelerations.

#### Current Behavior
> What is the current behavior?
> The robot stays as nothing as passed to it (if the gravity is zero). It falls if the gravity is -9.81.

#### Steps to Reproduce
> Please provide detailed steps for reproducing the issue.
1. In dart/example/operationspacecontrol, Change gravity to 0 (line 55)  in main file.
2. Change setForces to setAccelerations at line 104 in controller.cpp file .
3. Built and run the experiment. Once simulation starts, press C to rotate the circle.
4. Robot will stay as if nothing is passed to it.

#### Code to Reproduce
Code is already provided in operationalspacecontrol example of dart. Make the changes above and it should be fine.

jslee02, 2019-01-08:
Thanks for the report!

If you want to move the joints by acceleration, you need to change the actuation mode using [this function](https://github.com/dartsim/dart/blob/77acb65e467c311b48752e48604eb5cdf42cce52/dart/dynamics/Joint.hpp#L134) to `ACCELERATION`. The default actuator type is `FORCE`. With `FORCE` actuator type, the acceleration is overwritten by the forward dynamics algorithms (it computes acceleration given position, velocity, and force).

mamurtaza88, 2019-01-08:
Thank you for the quick response as well as fix. It got me working. It will be great if either it can be mentioned on dartwiki or fixed in the future releases.

mxgrey, 2019-01-08:
> It will be great if either it can be mentioned on dartwiki or fixed in the future releases.

This definitely isn't the first time this has caused confusion. I don't consider a bug that needs to be "fixed" because it's conceptually sound, but maybe we should add a warning to the documentation of `Joint::setAcceleration(~)`. Something to indicate that the acceleration value will get overwritten if the actuation mode is anything besides `ACCELERATION`.

jslee02, 2019-01-08:
@mxgrey Sounds great! [I added this to the todo list](https://github.com/dartsim/dart/projects/16#card-16274482).

Closing this issue as I believe this is resolved. Feel free to reopen otherwise!

closed 2019-01-08 by jslee02