dartsim/dart #1218: ContactConstraint.ContactWithKinematicJoint fails on Ubuntu 32-bit

jslee02, 2019-01-10:
### Bug Report
> Please answer the following questions for yourself before submitting an issue.
- [x] I checked the [documentation](http://dartsim.github.io/) and found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
* Select DART version: 6.7.0
* Select OS name and version name(or number): Ubuntu 16.04, 18.04, 18.10 (32-bit)
* Select compiler name and version number: GCC 5.4.0, 7.3.0, 8.2.0

#### Expected Behavior
Pass all the tests

#### Current Behavior
`ContactConstraint.ContactWithKinematicJoint` fails on Ubuntu 32-bit:
```
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from ContactConstraint
[ RUN      ] ContactConstraint.ContactWithKinematicJoint
/<<PKGBUILDDIR>>/unittests/unit/test_ContactConstraint.cpp:89: Failure
The difference between bodyNode2->getLinearVelocity()[0] and 0.1 is 0.00023997047433670093, which exceeds tol, where
bodyNode2->getLinearVelocity()[0] evaluates to 0.10023997047433671,
0.1 evaluates to 0.10000000000000001, and
tol evaluates to 0.0001.
/<<PKGBUILDDIR>>/unittests/unit/test_ContactConstraint.cpp:89: Failure
The difference between bodyNode2->getLinearVelocity()[0] and 0.1 is 0.00023996105705138826, which exceeds tol, where
bodyNode2->getLinearVelocity()[0] evaluates to 0.10023996105705139,
0.1 evaluates to 0.10000000000000001, and
tol evaluates to 0.0001.
[  FAILED  ] ContactConstraint.ContactWithKinematicJoint (34 ms)
[----------] 1 test from ContactConstraint (34 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (34 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] ContactConstraint.ContactWithKinematicJoint
```

#### Steps to Reproduce
> Please provide detailed steps for reproducing the issue.
1. cmake ..; make tests; ctest

jslee02, 2019-01-10:
I personally don't think there are many people using 32-bit systems, but let's fix this issue by relaxing the tolerance since it's a small change. And let's consider once it becomes more cumbersome to keep supporting 32-bit systems at the time.

By the way, I just wondered what would be a nice way to know whether the code is being built on 32-bit systems in compile time. Please let me know if you have any idea on this. :smile:

mxgrey, 2019-01-10:
I think you can use the variable [`CMAKE_SIZEOF_VOID_P`](https://cmake.org/cmake/help/v3.5/variable/CMAKE_SIZEOF_VOID_P.html) at configure time. Although that might not always be reliable, since someone could use compiler flags to force a 32 bit compilation on a 64 bit machine, and depending on how they do that, it might not be caught by `CMAKE_SIZEOF_VOID_P`.

jslee02, 2019-01-10:
Thanks! But I just realized that we already have the preprocessor for this, [`DART_ARCH_32BITS`](https://github.com/dartsim/dart/blob/master/dart/common/Platform.hpp#L43). :sweat_smile:

jslee02, 2019-01-10:
Fixed by #1219

PR #1219 (2019-01-10): Fix failure of ContactConstraint.ContactWithKinematicJoint on Ubuntu 32-bit
Fixes #1218 

***

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
closed 2019-01-10 by jslee02