dartsim/dart #1233: Compilation Failure when Including dart/gui/gui.hpp

acxz, 2019-02-03:
> If you are reporting a bug, please use the following template.
After upgrading to DART 6.7.2, I have been failing to compile codebases that include gui.
Specifically when including the below line:

    #include <dart/gui/gui.hpp>

The following error shows up:

    /usr/include/dart/gui/gui.hpp:7:10: fatal error: dart/gui/glut/glut.hpp: No such file or directory

I reinstalled `libdart6-all-dev` on Ubuntu 18 and it is the most recent version available from the PPA, however, the problem still persists. This was not an issue in 6.6.

I was wondering, if the glut folder inside dart/gui was forgotten for some reason.

If this could be issue could be fixed or if someone can guide me in the direction of creating working simulation code (maybe due to API changes in 6.7.2) that would be great.

jslee02, 2019-02-04:
It should be fixed by now. I'm closing issue for now but please reopen if the issue is persistent.

closed 2019-02-04 by jslee02

acxz, 2019-02-04:
Just confirmed it. Everything works as expected!!!

dt1729, 2023-10-31:
Working with 6.6.0 and still have this error:

`In file included from /usr/include/dart/utils/utils.hpp:12,
                 from /home/dt/QudrapedPapers/tutorials/tutorials/tutorial_biped/main.cpp:35:
/usr/include/dart/utils/XmlHelpers.hpp:240:10: fatal error: dart/utils/detail/XmlHelpers-impl.hpp: No such file or directory
  240 | #include "dart/utils/detail/XmlHelpers-impl.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
`

trying to compile tutorial_biped.

aliciaji1993, 2024-04-19:
Ran into the same issue when compiling gazebo from source with libdart 6.13:
[image]

libdart version info:
`apt-show-versions libdart6-all-dev
libdart6-all-dev:amd64/jammy 6.13.0-2118~202403250806~ubuntu22.04.1 uptodate`

Luckily, I tried to install libdart with homebrew before, and when I do file search I found the missing files in the brew directory:
[image]

Copied over the files and the compilation finished.

Hope this helps.

aliciaji1993, 2024-04-19:
> Working with 6.6.0 and still have this error:
> 
> `In file included from /usr/include/dart/utils/utils.hpp:12, from /home/dt/QudrapedPapers/tutorials/tutorials/tutorial_biped/main.cpp:35: /usr/include/dart/utils/XmlHelpers.hpp:240:10: fatal error: dart/utils/detail/XmlHelpers-impl.hpp: No such file or directory 240 | #include "dart/utils/detail/XmlHelpers-impl.hpp" | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ compilation terminated. `
> 
> trying to compile tutorial_biped. Ubuntu 20.04

commented above, hope this helps

dt1729, 2024-05-17:
I think I solved it similarly, but thanks on behalf of the community!

shailypriyanka, 2024-09-18:
> Ran into the same issue when compiling gazebo from source with libdart 6.13: ![Screenshot from 2024-04-19 12-16-33](https://private-user-images.githubusercontent.com/1229757/323831053-b6e403c0-a704-4723-9abb-c81540b8165c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2NjY5NjMsIm5iZiI6MTcyNjY2NjY2MywicGF0aCI6Ii8xMjI5NzU3LzMyMzgzMTA1My1iNmU0MDNjMC1hNzA0LTQ3MjMtOWFiYi1jODE1NDBiODE2NWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MThUMTMzNzQzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDRkZDMyYmNiOTQ1YWFkNTY3NTYxZDJlNzQwNzgzNmQwYTA3NzY5N2RjNmIyNWViYmJhYjI2NzhlNWVhYzE3NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.1uweTVa396LaVlrhnPkiJ3H0Vi2oWDB1oUz5eeHgJEg)
> 
> libdart version info: `apt-show-versions libdart6-all-dev libdart6-all-dev:amd64/jammy 6.13.0-2118~202403250806~ubuntu22.04.1 uptodate`
> 
> Luckily, I tried to install libdart with homebrew before, and when I do file search I found the missing files in the brew directory: ![Screenshot from 2024-04-19 12-18-58](https://private-user-images.githubusercontent.com/1229757/323831314-aeb48c2e-9e0e-4311-aeee-738b2cce7ee9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2NjY5NjMsIm5iZiI6MTcyNjY2NjY2MywicGF0aCI6Ii8xMjI5NzU3LzMyMzgzMTMxNC1hZWI0OGMyZS05ZTBlLTQzMTEtYWVlZS03MzhiMmNjZTdlZTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MThUMTMzNzQzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjIzMzY0YjQ1ZThlMjIxZDE0MmFhZTY4Mjk0OTVkMmY1NjNkMDBkZTMwZDQ3ZTQxZWMxZmQ2NzEyMGMyMDkyYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.SpljUFe8jDfCGejCiGEZzTtd9ArIh2WN4jGDa0jKgS0)
> 
> Copied over the files and the compilation finished.
> 
> Hope this helps.

I installed libdart using homebrew but unable to find those files showing in the dartsim/ part. 
I think it not at all installed at first.

mamurtaza88, 2024-10-30:
I was able to manually copy the file XmlHelpers-impl.hpp from dart/dart/utils/detail/ in source github code to /usr/include/dart/utils/detail/ . I had to manually create the detail folder in /usr/include/dart/ though but it resolves the issue. Apparently it is not created or copied when installing using apt install from package manager.