dartsim/dart #1234: Mesh - Plane collisions have caveats

mxgrey, 2019-02-08:
As we've been using dartsim in [ignition-physics](https://bitbucket.org/ignitionrobotics/ign-physics/), we've been finding that mesh-plane collisions need to meet a bunch of conditions to work properly, so I wanted to document that here.

The caveats depends on which collision detector is being used:

### FCL

dartsim doesn't support planes in FCL at all currently, so all `PlaneShape`s are automatically converted into thin boxes.

### Bullet

While Bullet seems to work okay for collisions between plane/primitive pairs, it seems to always fail between plane/mesh pairs. **Do not use the Bullet collision detector when you need meshes to collide with planes!**

### ODE

ODE seems to mostly work great for plane/mesh collisions, but for some reason it completely fails to detect collisions if the normal is facing directly along the X, Y, or Z axis (positively or negatively) while the offset is a negative value. I have no idea why that would be an edge case, but it happens consistently. If the normal vector is even microscopically rotated off one of those axes, it will detect collisions just fine with a negative offset.

So if you are using the ODE collision detector when colliding meshes with planes **try not use a negative offset for the `PlaneShape`!**.

I'll be submitting a PR to demonstrate these cases soon.

erwincoumans, 2019-02-14:
How are you representing concave triangle meshes in Bullet? The recommended way is to use a convex decomposition of the mesh (for example using VHACD) and store the convex parts in a btCompoundShape. That should work reliably. Many meshes are convex anyway, so they should use btConvexHull. Those convex meshes would need to be tagged as convex.

mxgrey, 2019-02-15:
Our current `MeshShape` class is meant to handle all categories of triangle meshes, not just convex meshes. There has been past discussion about the possibility of automatically decomposing concave meshes into sets of convex meshes so that Bullet can handle them better, but my personal feeling is that automatic convex decomposition is outside the scope of DART, and would be better left to a user to find an appropriate tool for that.

But this does emphasize the potential value of making convex meshes a first-class shape type in DART, which has been proposed before (#872).

erwincoumans, 2019-02-15:
So btGImpactMeshShape versus btStaticPlaneShape is not implemented/doesn't work in Bullet? I don't use btGImpactMeshShape, but it seems a rather easy thing to support collision against a plane. We could fix that on the Bullet side. How about filing an issue?

mxgrey, 2019-02-15:
Either that pair is not working as expected in Bullet, or there's a bug in DART's wrapper. It would probably be good to make a Bullet-only example of the failure to pinpoint whether it's a problem in Bullet (and if it turns out that the Bullet-only example works fine, then that'll be useful to indicate that we need to make a fix in how DART is using Bullet).