dartsim/dart #1247: CMake should fail if no collision library available

egordon, 2019-02-20:
Installing v6.7.0 on a new machine (Arch Linux) and forgot to install FCL.

CMake warned about missing FCL and ODE collision, but didn't exit with an error.

Running `make`, of course, failed.

I think it makes sense to have `cmake` exit with error if it know compilation is doomed to fail.

`make` output:
```
In file included from /home/egordon/Workspace/dart/dart/collision/fcl/BackwardCompatibility.cpp:33:
/home/egordon/Workspace/dart/dart/collision/fcl/BackwardCompatibility.hpp:49:10: fatal error: fcl/config.h: No such file or directory
 #include <fcl/config.h>
          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [dart/CMakeFiles/dart.dir/build.make:1493: dart/CMakeFiles/dart.dir/collision/fcl/BackwardCompatibility.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/egordon/Workspace/dart/dart/collision/fcl/FCLTypes.hpp:37,
                 from /home/egordon/Workspace/dart/dart/collision/fcl/FCLCollisionDetector.hpp:38,
                 from /home/egordon/Workspace/dart/dart/collision/fcl/FCLCollisionDetector.cpp:33:
/home/egordon/Workspace/dart/dart/collision/fcl/BackwardCompatibility.hpp:49:10: fatal error: fcl/config.h: No such file or directory
 #include <fcl/config.h>
          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [dart/CMakeFiles/dart.dir/build.make:1506: dart/CMakeFiles/dart.dir/collision/fcl/FCLCollisionDetector.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:342: dart/CMakeFiles/dart.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

`cmake` output:
```
-- ============================================
--                 DART 6.7.0
-- ============================================
-- Found CCD: /usr/include (found version "2.0") 
-- Could NOT find FCL (missing: FCL_INCLUDE_DIRS FCL_LIBRARIES) 
-- Found ASSIMP: /usr/include (found version "4.1.0") 
-- Performing Test ASSIMP_AISCENE_CTOR_DTOR_DEFINED
-- Performing Test ASSIMP_AISCENE_CTOR_DTOR_DEFINED - Success
-- Performing Test ASSIMP_AIMATERIAL_CTOR_DTOR_DEFINED
-- Performing Test ASSIMP_AIMATERIAL_CTOR_DTOR_DEFINED - Success
CMake Warning at cmake/DARTFindDependencies.cmake:103 (message):
  Octomap 1.9.0 is found, but Octomap 1.9.0 or greater is not supported yet.
  Please see 'https://github.com/dartsim/dart/issues/1078' for the details
Call Stack (most recent call first):
  CMakeLists.txt:137 (include)

-- Looking for ipopt - NOT found, to use dart-optimizer-ipopt, please install ipopt
-- Looking for nlopt - NOT found, to use dart-optimizer-nlopt, please install nlopt
-- Looking for pagmo - NOT found, to use dart-optimizer-pagmo, please install pagmo
-- Looking for ode - NOT found, to use dart-collision-ode, please install ode
-- Looking for Bullet - NOT found, to use dart-collision-bullet, please install libbullet-dev
-- Found FLANN: /usr/include (found version "1.9.1") 
-- Found TINYXML2: /usr/include (found version "7.0.1") 
-- Could NOT find OpenSceneGraph nor OpenThreads -- we will skip dart-gui-osg
If you believe you do have both OSG and OpenThreads installed, try setting OSG_DIR
-- Adding 8 components including the default component 'dart'
-- Adding 45 tests (comprehensive: 16, regression: 4, unit: 25)
-- Adding 16 examples
-- Adding 6 tutorials
```

mxgrey, 2019-02-20:
It's strange that configuration didn't quit with an error since FCL is marked as a required dependency [here](https://github.com/dartsim/dart/blob/e94a7ccec8a9b4d08a9e395c1df5fe7f0640bb2c/cmake/DARTFindfcl.cmake#L9).

I wonder if something is wrong with the way `find_package_handle_standard_args` is being used in [Findfcl.cmake](https://github.com/dartsim/dart/blob/e94a7ccec8a9b4d08a9e395c1df5fe7f0640bb2c/cmake/Findfcl.cmake#L60)?

jslee02, 2019-02-20:
> I wonder if something is wrong with the way `find_package_handle_standard_args` is being used in [Findfcl.cmake](https://github.com/dartsim/dart/blob/e94a7ccec8a9b4d08a9e395c1df5fe7f0640bb2c/cmake/Findfcl.cmake#L60)?

Yeah, you found it! This is because the package name (FCL in this case) doesn't respect the letter case, which it should do. Let me fix this.

PR #1250 (2019-02-21): Fix not failing for missing required dependencies
Fixes #1247
commit 0a1996ed (2019-02-21): Fix not failing for missing required dependencies (#1250)

Fixes #1247
closed 2019-02-21 by jslee02