dartsim/dart #1248: How to run headless simulation

adamconkey, 2019-02-21:
I'm new to working with the DART codebase, so forgive me if this is a stupid question. I have an application that doesn't require the OpenGL rendering offered by dartsim, and I am wondering how easy is it to run DART headless?

I want to use DART as my physics engine where I update the state over ROS and render the physics in rViz. All of the tutorial examples involve creating an OpenGL window and calling glutMainLoop(), which I find problematic because I don't know how to simultaneously be running a ROS subscriber at the same time. I appreciate any input on this.

adamconkey, 2019-02-21:
I think I may have answered my own question by looking through your regression tests. It seems like `world->step()` does the trick where `world` is a `WorldPtr`. I'll leave this open for a bit though if someone has better input on handling my use case.

mxgrey, 2019-02-21:
> It seems like `world->step()` does the trick where `world` is a `WorldPtr`

That's exactly right! Just remember that by default, all forces get cleared after each call to `World::step()`, so be sure to reapply any joint forces and applied external forces at each time step.

Also this might be a good opportunity to plug the recently created [DART Forum](https://dartsim.discourse.group/) where we encourage users to post questions about DART that aren't issues or bug reports.

closed 2019-02-21 by mxgrey

adamconkey, 2019-02-21:
Ah good to know about the forum! I knew this wasn't the best place to ask this question but wasn't sure where else to ask it. I will start using that more.

qleonardolp, 2025-01-28:
> That's exactly right! Just remember that by default, all forces get cleared after each call to `World::step()`, so be sure to reapply any joint forces and applied external forces at each time step. [...]

Is the Forum still online?

jslee02, 2025-01-28:
> Is this Forum still online?

It's not, unfortunately. Please use [Discussions](https://github.com/dartsim/dart/discussions) moving forward.