dartsim/dart #1251: Using the BodeNodeCollisionFilter

henryclever, 2019-02-22:
Hi, I'm trying to use the BodyNodeCollisionFilter to disable collision checking on certain bodynodes from my skeleton. I'm trying to build a binding for it in PyDart2, also. I have tried the following code:

```
auto collisionOption = world->getConstraintSolver()->getCollisionOption();
auto collisionFilter = collisionOption.getCollisionFilter();
collisionFilter->addBodyNodePairToBlackList(body0, body1);
```

However, it does not appear that getCollisionFilter is a valid function and I do not see it anywhere the Dart source code. How can I use the filter?

Thanks,
Henry C.

jslee02, 2019-02-22:
[`collisionFilter` is a (public) member of `CollisionOption`](https://dartsim.github.io/dart/v6.7.2/d4/d5b/structdart_1_1collision_1_1CollisionOption.html#a01d6425f781c72dc0da44de272b2ce34) so you can simply access it as:
```cpp
auto collisionFilter = collisionOption.collisionFilter;
```

As a note, we encourage users to use [DART Forum](https://dartsim.discourse.group/) to post questions that are not bug reports. Please use the forum for further questions, thanks!

closed 2019-02-22 by jslee02

henryclever, 2019-02-22:
Thanks for the help! I will be sure to do that in the future.

Henry C.