dartsim/dart #1256: Translation values are doubled when setting transform for BodyNode

adamconkey, 2019-03-01:
### Bug Report

I'm not certain this is a bug, but it seems bizarre to me nonetheless.

#### Environment
* DART version: release-5.1
* OS name and version name(or number): Ubuntu 16.04
* Compiler name and version number: GCC 5.4.0

#### Expected Behavior
When I create a BodyNode with a specified world transform, I expect the BodyNode to have that transform exactly when I inspect it after creation.

#### Current Behavior
When I create a BodyNode with a specified world transform, the BodyNode transform translation is **double** what I expect it to be, even though in the rendered world it makes sense. That is, everything looks okay in the world, but it's as if the position coordinates in the world are somehow half the "real world", and object positions are set with doubled coordinates. Object dimensions, however, seem normal and are not doubled.

#### Steps to Reproduce
1. Create an `Eigen::Isometry3d` transform with non-zero translation.
2. Set that transform in the `JointProperties being set for the BodyNode.
3. Inspect the transform of the BodyNode after creation.
4. The translation values should be doubled.

#### Code to Reproduce

    Eigen::Isometry3d tf = Eigen::Isometry3d::Identity();
    tf.translation()[0] = -0.4;
    tf.translation()[1] = 0;
    tf.translation()[2] = 0.1475;

    FreeJoint::Properties properties;
    properties.mName = "object_joint";
    properties.mT_ParentBodyToJoint = tf;
    properties.mT_ChildBodyToJoint = tf.inverse();

    dart::dynamics::BodyNodePtr bn = box->createJointAndBodyNodePair<JointType>(nullptr, properties).second;

    std::cout << "INPUT: \n"
              << tf.translation() << std::endl;
    std::cout << "BODY NODE: \n"
              << bn->getWorldTransform().translation() << std::endl;

I get the following output from this:
    
    INPUT: 
      -0.4
         0
    0.1475
    BODY NODE: 
     -0.8
        0
    0.295

jslee02, 2019-03-01:
This is expected behavior. The transform of a joint is a product of tf1 * tf2 * tf3^{-1} where
* tf1: `mT_ParentBodyToJoint`
* tf2: the transform by the joint value
* tf3: `mT_ChildBodyToJoint`.
Note that tf1 and tf3 are static transform while tf2 is a function of the joint values. tf1 and tf3 are offset to specify the transform of the child bodynode when when the tf2 is identity.

In your case, you might want to set either of tf1 or tf3 (not both). Either way, you could get the same result for the bodynode transform, but it will have different joint transform (the frame of the joint). In FreeJoint case, it doesn't matter anyway.

adamconkey, 2019-03-01:
Ah okay, I thought you always had to set both of those transforms based on [the tutorial](https://github.com/dartsim/dart/blob/release-5.1/tutorials/tutorialCollisions-Finished.cpp). But I guess in that tutorial there was a divide by 2 in the transform and then it was placing it in the middle of the object.

I tried only setting one of the transforms and you are right, I get the expected output. Thanks for your help!

closed 2019-03-01 by adamconkey