dartsim/dart #1261: Reference frame of nearest point

Macxou, 2019-03-13:
Hi, 
I'm currently using fcl 0.5 and i would like to compute the distance between 2 objects.
Both of them are placed in the world frame, so I can display them correctly (with VTK)

[image]
_The blue and red sphere are corresponding to the closest points given by the distance method.
I want to compute the distance between the tool and the join 4 which is the part on the left_

 But when I try to compute the distance between them, I get a wrong value and the closest points are not even the closest one. 
I guess that this is the closest one but they are given in the wrong frame. 
I 've seen the post : [https://github.com/dartsim/dart/issues/1102](url) where you explain that this issue is fixed in FCL 0.6.0 which is not released yet. 
I would like to know when FCL 0.6 will be released or how i can fix manually this issue.

Regards

mxgrey, 2019-03-13:
If the issue is fixed in FCL 0.6, then you can build and install that version of FCL from source and then also build and install dartsim from source.

Macxou, 2019-03-13:
Well, FCL 0.6 is not released yet ..
And my post is related to 
         - https://github.com/flexible-collision-library/fcl/issues/171
         - https://github.com/flexible-collision-library/fcl/pull/288

mxgrey, 2019-03-13:
If building against the master branch of FCL isn't an option for you, then you could look around through [these changes](https://github.com/flexible-collision-library/fcl/pull/288/files) to figure out what transformations you need to apply to the points to get them into the world frame. Just be careful if you do this, because you'll need to get rid of those transformations if you ever switch over to FCL 0.6.

If you want to know when FCL 0.6 will be released, then you'll have to ask on the FCL Github page. I think we'll be ready to support it whenever it drops.

Macxou, 2019-03-13:
Thank you so much

closed 2019-04-24 by jslee02