dartsim/dart #1264: How to use setPositions for closed loop chain constrained by BallJoint

Jaemin87, 2019-03-14:
I implemented the closed loop chain such as an example "RigidLoop". I implemented force-based control part using "setForce", however, I was trying to use "setPosition" but the ball joint constraint was violated and the chain blowed up. In detail, I did as follows: 
4-bar linkages: driving joint 1, passive joint 2 and 3, the ball joint constraint 4. and I update the joint position of the driving joint 1 in my worldnode. It is possible to update my kinematics using "setPosition" in this case?

mxgrey, 2019-03-15:
Using `setPosition(~)` in between simulation steps is strongly discouraged because it creates physical discontinuity in the simulation (calling `setPosition(~)` will effectively teleport the bodies in a way that defies physics). It's very normal for physical discontinuities to make things "blow up" as you describe, because they violate important assumptions that are needed by the math that governs the physics in the simulation.

You may want to consider using `joint->setActuatorType(dart::dynamics::Joint::SERVO);` and then you can use `joint->setCommand(desired_velocity);` to specify a velocity command for the joint. To emulate position control, you can choose a velocity that would drive the joint to the desired position in an acceptable amount of time.

Alternatively, you can use typical force control and drive towards the desired position using a traditional PID controller.

Jaemin87, 2019-03-15:
I understand. Thank you for your help.

closed 2019-03-15 by Jaemin87

jslee02, 2019-03-15:
Don't forget to use the [forum](https://dartsim.discourse.group/) for questions!