dartsim/dart #1267: URDF parser shouldn't treat multitree as an error

jslee02, 2019-03-29:
URDF parser errors when the root link is "world" (treated as a keyword) and the root link has multiple children. However, DART supports multitree in a skeleton. Moreover, this logic continues with this error and yet surprisingly it successfuly creates multitree anyway.

We should make it clear that it's not an error and update the logic accordingly.

mxgrey, 2019-04-01:
I think URDF wasn't designed to support multi-tree robot descriptions, so the original author of this (which wasn't me; all I did was refactor this code) decided that we should consider instances of multi-tree URDFs to be an error.

There are a few things about our URDF parser that aren't really standard, though, so I'm completely in favor of supporting a superset of the URDF standard.

jslee02, 2019-04-02:
> I think URDF wasn't designed to support multi-tree robot descriptions

That is true. So I think we should either return `nullptr` with an error message for that case (which is not the current implementation) or allow multi-tree robot with a warning message saying it's not the standard.

I'm inclined to the second option.

mxgrey, 2019-04-02:
The second option sounds good to me too :+1:

jslee02, 2019-04-06:
Resolved by #1270

closed 2019-04-06 by jslee02