dartsim/dart #1268: Allow Chain to include bodies have multiple children

jslee02, 2019-03-29:
[`Chain` doesn't allow including bodies that has multiple child bodies](https://github.com/dartsim/dart/blob/56205a31896592891ada2c8cd11079b151e8e8ef/dart/dynamics/Chain.hpp#L44-L45), but this criteria seem too strict. To me, a chain shouldn't have a branch in it but should be fine to be in a tree.

The current behavior prevents creating an proper IK from an end-effector if the parent body of end-effector has another child body: the IK will use only the parent joint of the end-effector but not the other ancestor joints to the root. This could be workarounded by manually specifying the desired DOFs to be used, but it's not ideal.

mxgrey, 2019-04-01:
The `Chain` class was specifically designed to grab portions of a Skeleton that contain no branching.

Why not use the [`Linkage` class](https://github.com/dartsim/dart/blob/56205a31896592891ada2c8cd11079b151e8e8ef/dart/dynamics/Linkage.hpp) for the use case that you're talking about?

If the problem is that `Linkage::Criteria` isn't convenient enough, we could give it a constructor to match the `Chain::Criteria` class, like:

```
Linkage::Criteria::Criteria(BodyNode* start, BodyNode* target, bool includeBoth = false);
```

and it'll construct a `Linkage` the same way it would construct a `Chain` but without the no-branching condition.

jslee02, 2019-04-02:
Yeah, `Linkage` can be used for my use case since `Chain` is a special case of `Linkage`. And the constructor looks great to me.

The reason I proposed the criteria relaxation (optionally by adding a flag) of `Chain` was that `Linkage` is to represent subtree (since it allows multiple targets) so it might be a bit too general for non-strict chains.

If we go with the option of not changing `Chain` class then it'd be good to add a function `Linkage::isChain()` that returns true when the number of targets is one.

mxgrey, 2019-04-02:
> If we go with the option of not changing Chain class then it'd be good to add a function `Linkage::isChain()` that returns true when the number of targets is one.

This is kind of a tricky semantic issue in my mind. The intention of the "chain" concept was to assure users that they've selected a subset of a skeleton where the selection can be abstracted as two transforms (the first transform in the chain and the last transform in the chain) without impacting anything outside of the selection.

In the case you're describing, the selection is a linear sequence, but there are bodies parallel to the sequence that are impacted by more than just the endpoints of the sequence.

What if the function were named `isSequence()` instead to distinguish it from the `Chain` concept?

jslee02, 2019-04-04:
I see your point. Let's keep the Chain as it is and give some utility functionalities to Linkage to support "sequence" structure in a convenient way.

> What if the function were named `isSequence()` instead to distinguish it from the `Chain` concept?

:+1:

PR #1273 (2019-04-09): Add a new Criteria constructor for creating sequence
### Problem
No convenient way to create a sequence Linkage.

### Solution
Add an utility constructor to `Linkage::Criteria` to create a sequence Linkage:
```cpp
Criteria(BodyNode* start, BodyNode* target, bool includeBoth = false);
```

### Issues
* Resolve #1268 

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
-  [... cut]
closed 2019-04-09 by jslee02