dartsim/dart #1287: HeightMapShape not implemented in SimWindow::drawShape

farg967, 2019-04-24:
I'm trying to implement a HeightMapShape as below..

 >  ...
>   BodyNode* bn = ground->createJointAndBodyNodePair<WeldJoint>().second;
> 
>   std::shared_ptr<HeightmapShape<double>> shape = std::make_shared<HeightmapShape<double>>();
> 
>   Eigen::Vector3d scale(1.0, 1.0, 1.0);
>   shape->setScale(scale);
>
>   // h is a matrixXd of elevations
>   // flatten h into vector
>   std::vector<double> heightsVec(h.data(), h.data() +h.cols()*h.rows());
> 
>   shape->setHeightField(h.cols(), h.rows(), heightsVec);
> ...

This seems to implement fine but I get the following error.

> Error [SimWindow.cpp:474] [SimWindow::drawShape] Attempting to draw an unsupported shape type [HeightmapShape (d)].

This is due to HeightMapShape not being implemented in SimWindow::drawShape.

Is there a different way HeightMapShape should be used? Or is this just not implemented yet?

jslee02, 2019-04-24:
HeightMapShape visualization is not supported at this moment, but the collision checking and physics simulation with heightmap should work.

jslee02, 2019-04-24:
We don't have plan to improve SimWindow but [will add heightmap support to the OSG renderer](https://github.com/dartsim/dart/issues/1289).

jslee02, 2019-04-24:
Closing this in favor of https://github.com/dartsim/dart/issues/1289 since this is not a bug but about missing feature.

closed 2019-04-24 by jslee02