dartsim/dart #1299: Source Code Compilation Re: `pagmo`

acxz, 2019-04-30:
#### Environment
* DART version: 6.8.2
* OS name and version name(or number): Ubuntu 18.04.2
* Compiler name and version number: GCC 7.4.0
#### Expected Behavior
The expected behavior is for DART to successfully compile with the optional dependency of `pagmo`.

#### Current Behavior

When `pagmo` is not installed, cmake executes fine. ([`cmake_out_no_pagmo.txt`](https://gist.github.com/acxz/f9bb7aec03cd4a790ac4b7262aaceacd#file-cmake_out_no_pagmo-txt))

After installing `pagmo`, cmake returns multiple errors. Notice `pagmo` is found successfully. There seems to be an error with referencing `ipopt`. ([`cmake_out_with_pagmo.txt`](https://gist.github.com/acxz/f9bb7aec03cd4a790ac4b7262aaceacd#file-cmake_out_with_pagmo-txt))

#### Steps to Reproduce
1. Install `pagmo` via instructions [here](https://esa.github.io/pagmo2/install.html)
    Make sure to enable the `PAGMO_WITH_{EIGEN3,NLOPT,IPOPT}` cmake options.
    Essentially:
    ```
    git clone https://github.com/esa/pagmo2
    cd pagmo2
    git checkout tags/v2.10
    mkdir build
    cd build
    cmake -DPAGMO_WITH_EIGEN3=ON -DPAGMO_WITH_NLOPT=ON -DPAGMO_WITH_IPOPT=ON ..
    sudo make install
    ```
2. Clone and Build DART

#### Code to Reproduce
None needed.

jslee02, 2019-05-01:
Thanks for the report! I could reproduce this error, and it should be fixed by #1301.

closed 2019-05-01 by jslee02

acxz, 2019-05-01:
Thanks for the fix! Just wanted to pipe in, build is now successful for me.

PR #1301 (2019-05-01): Fix building with pagmo's optional dependencies
Resolves #1299 

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change