dartsim/dart #1304: Proposal: Tweak the examples directory to make dart::gui::osg more accessible

mxgrey, 2019-05-03:
Many new users will go to the `dart/examples` directory to get started with writing an application, but if they click randomly (which many are likely to do), odds are they'll select one of the old glut-based examples instead of one of the newer osg-based examples.

I would propose changing the directory layout to something like

```
examples
|-- glut_examples_(deprecated)
|   |-- addDeleteSkels
|   |-- atlasSimbicon
|   |-- ...
|-- osg_examples_(recommended)
    |-- osgAtlasPuppet
    |-- osgAtlasSimbicon
    |-- ...
```

We may also want to make sure that all the old glut-based examples have been ported over to osg. If we can put ourselves in a position where people only expect support for OSG instead of both OSG and glut, it might make it easier to address questions.

jslee02, 2019-05-03:
I like this idea. One thing I'd like to add is changing the name of directories (so the executable names as well). As the intention is to imply the OSG is the recommended renderer, I would remove the `osg` prefix from the OSG examples and add `glut` prefix to GLUT examples. So the structure would be like:
```
examples
|-- glut_examples_deprecated
|   |-- glutAddDeleteSkels
|   |-- glutAtlasSimbicon
|   |-- ...
|-- osg_examples
    |-- atlasPuppet
    |-- atlasSimbicon
    |-- ...
```

Also, I would like to be consistent in the directory name case pattern to be either of all lower cases or Pascal style. I know we've been using Pascal style, but I'm slightly inclined to lower cases since it seems to be more widely used.

mxgrey, 2019-05-03:
That all sounds good to me!

PR #1305 (2019-05-06): Update examples directory to make dart::gui::osg more accessible
This PR tweaks the examples directory to make `dart::gui::osg` more accessible. Following changes are also made:
* Change example and tutorial directory names to all lower cases
* Remove `InSourceBuild.cmake` files in favor of `dart_build[_example/_tutorial]_in_source()` macros
* Format examples and tutorials code
* Add `hello_world` example

Resolves #1304

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set [... cut]
closed 2019-05-06 by jslee02