dartsim/dart #1308: DART_INCLUDE_DIRS doesn't include the dependent include directories

jslee02, 2019-05-06:
### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: 6.8.4 (installed from PPA)
* OS name and version name(or number): Ubuntu 16.04
* Compiler name and version number: GCC 5.4.0

#### Expected Behavior
`DART_INCLUDE_DIRS` should include all the include paths of the DART dependencies.

#### Current Behavior
`DART_INCLUDE_DIRS` only includes `/usr/include`

#### Possible Solution
We should either fix this or drop supporting CMake variables in favor of imported targets (related issue: https://github.com/dartsim/dart/pull/1209).

mxgrey, 2019-05-07:
Since `DART_LIBRARIES` is already being stuffed the imported targets, the user should get all the headers they need just by linking their target to `DART_LIBRARIES`. They shouldn't even need to explicitly use `DART_INCLUDE_DIRS` at all anymore.

Do you have a situation where the configuration/build is actually failing?

jslee02, 2019-05-07:
> Since `DART_LIBRARIES` is already being stuffed the imported targets, the user should get all the headers they need just by linking their target to `DART_LIBRARIES`. They shouldn't even need to explicitly use `DART_INCLUDE_DIRS` at all anymore.

Yeah, that is what I expect as well.

> Do you have a situation where the configuration/build is actually failing?

I don't have, but one user reported that. I'm clarifying if the user linked the imported targets using `DART_LIBRARIES`. In any case, it would be good to either fix `DART_INCLUDE_DIRS` and support it; or remove it entirely and encourage users to use `DART_LIBRARIES` instead.

mxgrey, 2019-05-07:
If we're going to choose between fixing it and removing it, I would remove it since it doesn't accomplish anything.

The one catch with removing it is that some users might be confused when it shows up completely blank, and might be misled into thinking that something in their configuration or installation is broken.

jslee02, 2019-05-07:
Good point. I meanted making `DART_INCLUDE_DIRS` as an undefined variable, but it seems CMake teats an undefined variable as an empty variable in `target_link_libraries` sliently. The user can be confused unless they checks if the variable is defined using `if(DEFINED ...)`.

closed 2025-11-02 by jslee02