dartsim/dart #1310: DART source install fails with pagmo built without eigen

acxz, 2019-05-07:
#### Environment
* DART version: 6.8.4
* OS name and version name(or number): Arch Linux (Installed from source, results should be the same on other Linux platforms.
* Compiler name and version number: GCC 8.3.0

#### Expected Behaviour
The expected behaviour is for DART to successfully compile with the optional dependency of `pagmo`. (Or maybe it is for DART to compile with pagmo built with Eigen, if so it should probably be caught in the CMake)

#### Current Behavior

When `pagmo` is installed without Eigen support, the CMake process of DART finishes, but the make process fails. Link to output: [link](https://gist.github.com/acxz/716465d36dc0e11a646ca5abf2d1b52d).

Basically it seems like how DART uses pagmo assumes pagmo to be built with eigen support but DART's cmake does not check for that, which causes errors during the build process.

#### Steps to Reproduce
1. Install `pagmo` via instructions [here](https://esa.github.io/pagmo2/install.html)
    Essentially:
    ```
    git clone https://github.com/esa/pagmo2
    cd pagmo2
    git checkout tags/v2.10
    mkdir build
    cd build
    cmake ..
    sudo make install
    ```
2. Clone and Build DART
    For this I just installed the required dependencies excluding pagmo. I don't think installing DART with the other optional dependencies would change this issue.

#### Code to Reproduce
None needed.

jslee02, 2019-05-07:
It seems this is a pagmo's issue. [This line](https://github.com/esa/pagmo2/blob/master/include/pagmo/pagmo.hpp#L61) should be replaced by:
```cpp
#if defined(PAGMO_WITH_EIGEN3)
#include <pagmo/algorithms/xnes.hpp>
#endif
```

acxz, 2019-05-07:
Ahh, I see. Confirmed the fix.

You're amazing @jslee02! Thanks for always being on top of things. This project is really in good hands!

bluescarni, 2019-05-07:
@jslee02 thanks for catching the issue! The fix will be in the next pagmo release.

bluescarni, 2019-05-07:
@jslee02 question: are you actually using the ``pagmo.hpp`` header in your code?

jslee02, 2019-05-07:
@bluescarni Yes, I'm using the header [here](https://github.com/dartsim/dart/blob/cfb60934880282e40d2448c040f9fbd45220dfc5/dart/optimizer/pagmo/PagmoMultiObjectiveSolver.hpp#L37).

bluescarni, 2019-05-07:
@jslee02 we are almost finished working on a PR esa/pagmo2#278 which turns pagmo into a compiled library. One of the effects of this change is to decrease the compile time of an empty C++ file which just includes ``pagmo.hpp`` from 1 minute to something like 3 seconds. If you ever noticed long compile times when including pagmo, the situation should get much better soon.

jslee02, 2019-05-07:
Sounds awesome! Yeah, I noticed the compile time increase when building an unit test including `pagmo.hpp`. Maybe I could reduce the compile time by including specific headers that I acutally need, but the change will resolve it in a better way. Look forward to the PR is merged!

jslee02, 2019-06-06:
Closing since https://github.com/esa/pagmo2/pull/281 is merged

closed 2019-06-06 by jslee02