dartsim/dart #1331: Two memory leaks in DART::gui::osg::Viewer

Aneoshun, 2019-05-21:
### Bug Report
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
* DART version: [master (also tested on release-7.0)]
* OS name and version name(or number): [Ubuntu 18.04]
* Compiler name and version number: [GCC 7.3.0]

#### Expected Behavior
Creating a dart::gui::osg::Viewer should not cause memory leak. 

#### Current Behavior
Currently, dart::gui::osg::Viewer causes two (identified) memory leaks.

#### Steps to Reproduce

1. Use the following code to reproduce the problem:
```
#include <iostream>
#include <memory>
#include <dart/gui/osg/osg.hpp>

int main(int argc, char **argv)
{
  for(size_t i=0; i<1000;i++){
    osg::ref_ptr<dart::gui::osg::Viewer> osg_viewer = new dart::gui::osg::Viewer;
  }
  return 0;
}
```
2. If you run this with `valgrind --tool=massif` you will see that the memory is not released after each iteration. 
Here is the result of massif: 
[image]
Please note that it uses 30MB of RAM to do strictly nothing. 

#### The leaks that I identified and how to fix them
The first leak comes from the `SaveScreen` callback in dart::gui::osg::Viewer.cpp (instanciated [here](https://github.com/dartsim/dart/blob/b6d8fa961dd191a5670fd2dc8073cb1a1cc54c7b/dart/gui/osg/Viewer.cpp#L222))
The issue is that this callback is registered in the camera, but this callback also keeps a pointer of the camera (`mCamera`). This creates a circular reference and breaks the garbage collection of OSG (both of these objects have always a pointer pointing to them).
To solve this, the destructor of the Dart viewer should call:
getCamera()->setFinalDrawCallback(0);

The second leak is caused by the way the WorldNodes are stored in the dart::gui::osg::Viewer, which relies on a std::map<WorldNode*,bool> mWorldNodes. This uses standard pointers and thus breaks the garbage collection of OSG. 
In my tests, I solve this issue by manually removing the only WorldNode that I am using via:
`_osg_viewer->removeWorldNode(_osg_world_node);` however this will not work when multiple WorldNodes are used (I am personally never in this situation).  
I see two possibles approaches to properly fix this second leak:
1. replace `WorldNode*` by `osg::ref_ptr<WorldNode>` this should enable OSG's garbage collection to work properly. 
2. Otherwise, you can implement a "clearWorldNodes" method that removes all the worldNodes in the destructor of the viewer. 

Sorry, the example above does not cover this second leak as it requires to add a WorldNode, which quickly makes the example quite bigger. However, I can potentially give you an example for this as well if really necessary. 

Here is the result of massif on the example listed above after fixing the code (note the 98% memory reduction...) : 
[image]

I hope this will help you to fix these leaks. 

Best,

jslee02, 2019-05-25:
Appreciated for the dedicated investigation on the issue. Let me look into this issue.

jslee02, 2019-06-03:
I could reproduce the memory leak cases. They should be fixed in #1349 by removing the circular dependency and using OSG reference counting pointer.

Aneoshun, 2019-06-03:
Excellent! Thank you.

PR #1349 (2019-06-05): Fix memory leaks from dart::gui::osg::Viewer (#1331)
This PR fixes #1331.

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [ ] Add unit test(s) for this change
closed 2019-06-05 by jslee02