dartsim/dart #1365: Could not find a configuration file for package "urdfdom_headers" that is   compatible with requested version "".

dongfangliu, 2019-07-05:
I currently tried to install DART on windows 10 .
I followed the [guide](https://dartsim.github.io/install_dart_on_windows.html) all the way through, using fresh new vcpkg and install all dependencies.

However, things stack on the configure process of cmake.

It always jumped out following message.
-----------------------------------------------------------------------------------------------------
Could not find a configuration file for package "urdfdom_headers" that is
  compatible with requested version "".

  The following configuration files were considered but not accepted:

    F:/vcpkg/installed/x64-windows/share/urdfdom_headers/urdfdom_headers-config.cmake, version: 1.0.4 (64bit)`
-----------------------------------------------------------------------------------------------------

Which i think it's reason to cause following generate failure if above warning is negelectd
--------------------------------------------------------------------------
CMake Error in dart/utils/urdf/CMakeLists.txt:
  Imported target "urdfdom" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.

--------------------------------------------------------------------------
I really confused, if there's no requested verison ,why not accepted?

jslee02, 2019-07-06:
I could reproduce this on the [CI build test](https://dev.azure.com/dartsim/dart/_build/results?buildId=1375). Let me take a look this.

dongfangliu, 2019-07-07:
> I could reproduce this on the [CI build test](https://dev.azure.com/dartsim/dart/_build/results?buildId=1375). Let me take a look this.

Cool, waiting for your replay

jslee02, 2019-07-07:
The root cause of the error is that `urdfdom_INCLUDE_DIRS` includes `/include` that is generated by the [`urdfdom-config.cmake.in`](https://github.com/ros/urdfdom/blob/master/cmake/urdfdom-config.cmake.in#L6) as you described in the post. I'm not sure if this an issue of `urdfdom` itself or the vcpkg package. What DART can do would be excluding `/include` from `urdfdom_INCLUDE_DIRS` (or any path begin with `/`), but it might lead to compilation errors because `tinyxml` headers are required to build with `urdfdom`.

I think this should be fixed either by the upstream or in the vcpkg package. But, unfortunately, we currently don't have the resources to fix dependency packages to support Windows at the moment. I've tried to fix this issue within DART but couldn't come up with a good solution.

dongfangliu, 2019-07-08:
Ok,thank you anyway

dongfangliu, 2020-02-15:
Now i find the problem.

It dues to the config file  `urdfdom-config.cmake` in `vcpkg-2020.01\installed\x64-windows\share\urdfdom`.

The 4th line should be  corrected to 
`set(urdfdom_INCLUDE_DIRS "${CMAKE_CURRENT_LIST_DIR}/../../include" )`

jslee02, 2020-04-27:
Thank you for looking into the problem. It sounds like the tinyxml location is different when urdfdom is built and DART is built. A potential fix would be change [line 4 of `urdfdom-config.cmake.in`](https://github.com/ros/urdfdom/blob/0da4b20675cdbe14b532d484a1c17df85b4e1584/cmake/urdfdom-config.cmake.in#L6) to use [`find_dependency()`](https://cmake.org/cmake/help/latest/module/CMakeFindDependencyMacro.html) so that tinyxml is always correctly found when `urdfdom-config.cmake` is loaded.

Closing as it looks like this issue should be fixed by urdfdom.

jslee02, 2020-05-02:
Reported to urdfdom: https://github.com/ros/urdfdom/issues/140

PR #1456 (2020-05-03): Add workaround for invalid path in finding urdfdom on Windows
Fixes #1365
closed 2020-05-03 by jslee02