dartsim/dart #1370: Simulation World resetCommand argument

costashatz, 2019-07-12:
> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

### Bug Report

- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment

* DART version: master
* OS name and version name(or number): Ubuntu 18.04
* Compiler name and version number: GCC 7.4.0

#### Expected/Weird Behavior

I am using a `World` to simulate a world using DART. When I do a `step()` I pass the `resetCommand = false`. Usually everything works fine until I used a soft body. In that case the behavior that I am getting when resetting or disable the rest of the commands in the step is different. In short, I get normal behavior when `resetCommand = true` and weird behavior otherwise.

#### Steps to Reproduce

1. Create a world with a soft body (a soft cube for example) and a rigid floor
2. Put the soft body at a height (e.g., 0.5)
3. Run the simulation with `resetCommand = true` and with `resetCommand = false`

Why I am getting this behavior? Any ideas?

Thanks a lot!

jslee02, 2019-07-13:
Thank you for the bug report. I could reproduce the issue from `soft_bodies` example with `resetCommand = false` option. Let me take a look if it can be quickly resolved.

jslee02, 2019-07-13:
This should be fixed by #1372! Let me know if it doesn't work for you.

costashatz, 2019-07-14:
> This should be fixed by #1372! Let me know if it doesn't work for you.

Seems to be working! Thanks a lot!

PR #1372 (2019-07-14): Reset internal force of point mass before every simulation step
The soft body simulation behaves unexpectedly when the simulation option is `resetCommand = false`. The root cause was that the internal forces of point masses, which are used for soft body simulation, are not reset to zero in every simulation step. In the rigid body case, this issues doesn't happen because the internal forces are set to the control command. Soft body, however, doesn't have control command API so it keeps the previous value, and this leads to the unexpected behavior.

This PR fixes the issue by always setting the  [... cut]
closed 2019-07-14 by jslee02