dartsim/dart #1393: [dartpy] Allow setting background color for OSG windows

chhinze, 2019-08-15:
**Problem**
In DART, the background color of an `dart::gui::osg::Viewer` can be set in the constructor with a `::osg::Vec4` vector. This behaviour is also exposed in the python bindings, but the  type `::osg::Vec4` is not defined. As far as I see, there is curently no possibility to set a background color.

**Solution**
Adding an `__init__` function binding for `dartpy.gui.osg.Viewer` and `dartpy.gui.osg.ImGuiViewer` with an `Eigen::Vector4f` would allow to set the color either with a numpy array or even via a list with 4 entries.
I already wrote a solution for myself (see https://github.com/dartsim/dart/compare/master...chhinze:fix/pyColorSupportOsgViewer) and would like to merge it, if desired.

**Alternatives**
There may be some OSG python binding projects, but I would prefer reducing the number of required Python packages.

jslee02, 2019-08-15:
The solution sounds good to me.

Indeed, I'd prefer minimizing the third-party library APIs to the public DART APIs for both C++ and Python. In that sense, I'd add a new constructor of `dart::gui::osg::Viewer` taking `Eigen::Vector4f` (and `setClearColor(const Eigen::Vector4f)`, and update the Python bindings accordingly.

> There may be some OSG python binding projects, but I would prefer reducing the number of required Python packages.

Agreed for the same reason above.

jslee02, 2019-08-17:
I believe this is resolved by #1393. Thanks for the fix @chhinze!

closed 2019-08-17 by jslee02

chhinze, 2019-08-17:
Thank you for merging it.