dartsim/dart #1394: Standalone Kinematics/Dynamics Library

fishyai, 2019-08-15:
**Is your feature request related to a problem? Please describe.**
DART has super fast kinematics calculations that I'd like to be able to use in my project--namely the URDF parsing and jacobian/end-effector pose calculation.

**Describe the solution you'd like**
It's be great if the library was organized as submodules for different aspects, then it'd be easier to just include the aspects that are super relevant.

**Describe alternatives you've considered**
I tried linking against the whole library, but it's big and has caused some other issues (I'll post about those separately). I also starting ripping out the pieces myself, but I don't have full familiarity with the library, so it's hard to do practically.

jslee02, 2019-08-15:
DART consists of the core library (kinematics, dynamics, collision detection) and optional components (gui, optionall collision detectors). We try to keep the core library to be light as much as possible with minimum dependencies. The components are only built when their dependencies installed in the system. I'd be happy to help to resolve the linking issues you have.

mxgrey, 2019-08-16:
I agree that the `dynamics` namespace has gotten rather bloated, containing content ranging from kinematics to dynamics to geometry.

I don't think it's feasible to fix this in dartsim-6, but for dartsim-7 I think it would be worthwhile to consider refactoring `dynamics` into `kinematics`, `dynamics`, and `geometry` libraries.