dartsim/dart #1420: Failure on compliation with dart.hpp

dongfangliu, 2019-09-23:
**Problem:**

when compling dart.hpp with my project, i face some error that i failed to remove.
It says

```
/usr/local/include/dart/dynamics/GenericJoint.hpp:68:3: error: invalid use of 'class std::set<typename dart::dynamics::GenericJoint< <template-parameter-1-1> >::Aspect>'
   DART_BAKE_SPECIALIZED_ASPECT_IRREGULAR(
   ^
In file included from /usr/local/include/dart/common/Composite.hpp:204:0,
                 from /usr/local/include/dart/common/common.hpp:6,
                 from /usr/local/include/dart/dart.hpp:34,
                 from /root/Skeletal/src/DartWrapper/mlSkeletonDynamics.h:2,
                 from /root/Skeletal/src/test2.cpp:3:
/usr/local/include/dart/dynamics/GenericJoint.hpp: In member function 'void dart::dynamics::GenericJoint< <template-parameter-1-1> >::setGenericJointAspect(std::unique_ptr<typename dart::dynamics::GenericJoint< <template-parameter-1-1> >::Aspect>&&)':
```
My environment is with cmake-3.15, gcc-5.1.0,g++-5.1.0. 
It's worth mentioned that i succeeded in building and installing dart.
Problem occurs when using dart in my project.
 
I took this error as c++/c standard related problem. So i add the requirement to std11.But that does not work.

**Hope anybody give a solution.**

-----------------------------------------------------------------------------------------------------------------

If not mind,you can check my CMakeLists.txt
```
cmake_minimum_required (VERSION 3.5.1)

project(skeletal  LANGUAGES CXX CUDA)
add_compile_options(-std=c++11)
set(CMAKE_C_STANDARD 11)

 
#cuda
enable_language(CUDA)
if(NOT DEFINED CMAKE_CUDA_STANDARD)
    set(CMAKE_CUDA_STANDARD 11)
    set(CMAKE_CUDA_STANDARD_REQUIRED ON)
endif()
include_directories(${CUDA_INCLUDE_DIRS})
include_directories(${CMAKE_CUDA_TOOLKIT_INCLUDE_DIRECTORIES})
set(CMAKE_CUDA_SEPARABLE_COMPILATION OFF)
set(CUDA_LINK_LIBRARIES_KEYWORD PUBLIC)
# OGRE AND IMGUI
find_package(OGRE 1.10 REQUIRED)
find_package(Freetype QUIET)

# the search paths
include_directories(${OGRE_INCLUDE_DIRS})
link_directories(${OGRE_LIBRARY_DIRS})
file(COPY ${OGRE_CONFIG_DIR}/resources.cfg DESTINATION ${CMAKE_BINARY_DIR})

set(IMGUI_SRCS 
    ${CMAKE_SOURCE_DIR}/3rdParty/ogre-imgui/imgui/imgui.cpp 
    ${CMAKE_SOURCE_DIR}/3rdParty/ogre-imgui/imgui/imgui_draw.cpp
    ${CMAKE_SOURCE_DIR}/3rdParty/ogre-imgui/imgui/imgui_widgets.cpp)
set(OGRE_IMGUI_SRCS ${CMAKE_SOURCE_DIR}/3rdParty/ogre-imgui/ImguiManager.cpp)
if(FREETYPE_FOUND)
    include_directories(${CMAKE_SOURCE_DIR}/3rdParty/ogre-imgui/imgui/misc/freetype/)
    list(APPEND IMGUI_SRCS ${CMAKE_SOURCE_DIR}/3rdParty/ogre-imgui/imgui/misc/freetype/imgui_freetype.cpp)
endif()

add_definitions(-DIMGUI_DISABLE_OBSOLETE_FUNCTIONS)
include_directories(${CMAKE_SOURCE_DIR} ${CMAKE_SOURCE_DIR}/3rdParty/ogre-imgui/imgui/ ${CMAKE_SOURCE_DIR}/3rdParty/ogre-imgui/)
add_library(OgreImgui ${IMGUI_SRCS} ${OGRE_IMGUI_SRCS})
set_property(TARGET OgreImgui PROPERTY POSITION_INDEPENDENT_CODE ON)
target_link_libraries(OgreImgui PUBLIC ${OGRE_LIBRARIES})

if(OGRE_Bites_FOUND)
    target_compile_definitions(OgreImgui PRIVATE -DHAVE_OGRE_BITES)
endif()

if(FREETYPE_FOUND)
    target_compile_definitions(OgreImgui PRIVATE -DUSE_FREETYPE)
    target_include_directories(OgreImgui PRIVATE ${FREETYPE_INCLUDE_DIRS})
    target_link_libraries(OgreImgui PRIVATE ${FREETYPE_LIBRARIES})
endif()

## ASSIMP 
find_package(ASSIMP)
## Eigen
find_package(Eigen3)
include_directories(${EIGEN3_INCLUDE_DIR})
##DART
set(DART_components utils-urdf gui)
set(DART_libraries dart dart-utils-urdf dart-gui)
find_package(DART 6.6.0 REQUIRED COMPONENTS ${DART_components} CONFIG)
include_directories(${DART_INCLUDE_DIRS})

include_directories(3rdParty)
#add all my files to a lib
file(GLOB_RECURSE CU_SRCS "cslbm/*.cu"  "cslbm/*.cpp"  "cslbm/*.h")

aux_source_directory(MeshLoad MY_SKELETON_SRCS)
aux_source_directory(DartWrapper MY_SKELETON_SRCS)

# add the source files as usual
add_executable(test test2.cpp ${MY_SKELETON_SRCS} ${CU_SRCS})
# this also sets the includes and pulls third party dependencies
target_link_libraries(test PUBLIC OgreImgui ${ASSIMP_LIBRARIES} ${DART_libraries})

find_package( OpenMP REQUIRED)
if(OpenMP_CXX_FOUND)
message("OPENMP FOUND")
target_link_libraries(test  PUBLIC OpenMP::OpenMP_CXX)
endif()

```

dongfangliu, 2019-09-24:
sorry, it's due to file confliction.

closed 2019-09-24 by dongfangliu

jslee02, 2019-09-24:
Glad you resolved the problem! Your `CMakeLists.txt` looks good to me. You may be already aware of it, but I'd like to mention that the [examples](https://github.com/dartsim/dart/tree/07fd1a15a9dfd16d71903baea7ed515a7106924d/examples) in DART have a `CMakeLists.txt` file in each  example folder so you can start from there.