dartsim/dart #1423: Switch to C++14 not mentioned in changelog

peci1, 2019-10-09:
### Bug Report
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
* DART version: 6.10.0
* OS name and version name(or number): Ubuntu 18.04
* Compiler name and version number: GCC 7.4.0

#### Expected Behavior
People should really get notified if you do such a big change as C++ standard upgrade is. It seems this change probably [broke Gazebo build](https://bitbucket.org/osrf/gazebo/issues/2663/build-error-gazebo9-and-dart), which builds with c++11 flag.

#### Current Behavior
Commit https://github.com/dartsim/dart/commit/a12231e92bc57623cde85afc6d5b2ef683f94bc8 brought in the new requirement for building with C++14 standard, but it is not mentioned in the changelog.

#### Steps to Reproduce
N/A

#### Code to Reproduce
N/A

scpeters, 2019-10-09:
Yeah, I wouldn't expect a minor release to change the required c++ standard.

jslee02, 2019-10-09:
The required C++ standard was changed as we stop supporting Ubuntu 14.04 because the compilers of all the platforms DART supports completely support C++14. But maybe that wasn't a good idea.

I think we can revert the change if this cause serious problem to the downstream users. The only new C++14 features used in DART is `std::make_unique` IIRC.

Before making that decision, I wonder if there is any chance that Gazebo 9 allows to build with C++14. This is because Gazebo 9 also supports Ubuntu >= 18.04 according to the project status in the [website](http://gazebosim.org).

In any case, I think the Gazebo build error shouldn't happen (or at least it should be detected the required C++ standard error earlier) if the required compiler options for DART were added when linking to DART. It can be done by either linking DART using DART target or using `DART_DEFINITIONS`, which is not currently provided by `DARTConfig.cmake`. This should be the same with C++11 standard so let me add `DART_DEFINITIONS` for this.

scpeters, 2019-10-09:
I don't think the problems are huge actually; ign-math4 also requires c++14; I'm not sure why gazebo suddenly started failing now over this change.

mxgrey, 2019-10-10:
I'm kind of suspicious of [this block](https://github.com/dartsim/dart/blob/1db05b083912e9f900b53c1b61aa865c6085c106/dart/CMakeLists.txt#L147-L151) in our CMakeLists.txt.

If I'm not mistaken, it's possible for cmake to ignore the public `-std=c++14` compile flag based on what cmake believes the necessary compile features are.

I think we should consider changing [this line](https://github.com/dartsim/dart/blob/1db05b083912e9f900b53c1b61aa865c6085c106/dart/CMakeLists.txt#L148) to use `target_compile_features` like the other line, but specify a compile feature that we know will mandate the use of C++14.

j-rivero, 2019-10-10:
> I don't think the problems are huge actually; ign-math4 also requires c++14; I'm not sure why gazebo suddenly started failing now over this change.

That could probably come for the fact that current compilers in gazebo9 accepted platforms are able to compile ignition-math4 simply by using the `c++11` flag or by the fact that `c++14` features are not present in the code used by gazebo (mostly the headers). My first experiment was to modify the standard to `c++11` using `Ubuntu Bionic gcc 7.4.0-1ubuntu1~18.04.1` and compile `ignition-mat4` branch. It compiles just fine here.

jslee02, 2020-03-01:
I have some updates on this issue and #1429:

From my [experiments](https://github.com/dartsim/dart/tree/archilinux/cpp14), I confirmed that `this block` in our `CMakeLists.txt` correctly passes necessary compile options when linking to DART by using `target_link_libraries(gazebo ${DART_LIBRARIES})`. So we don't need to introduce `DART_DEFINITIONS` additionally.

The problem is that the C++ standard is overridden to C++11 by this line in Gazebo, which prevents us to build C++14 code. For example, if [this line](https://github.com/dartsim/dart/blob/archilinux/cpp14/gazebo/CMakeLists.txt#L5) is disabled, (fake) Gazebo project can build with C++14 code (since it links to `DART_LIBRARIES`). Without the line, Gazebo libraries that don't depend on DART still should be able to build C++11 code by [this block](https://github.com/dartsim/dart/blob/archilinux/cpp14/gazebo/CMakeLists.txt#L22-L26).

costashatz, 2020-03-01:
> One thing weird to me is that this issue only happens only on archlinux (docker) but not Ubuntu.

My guess would be because of the CMake version? Which CMake version did you use in Ubuntu? Archlinux always uses the latest release..

Any potential fix for this?

jslee02, 2020-03-01:
2.8.6 for Gazebo, 3.5.1 for DART. I posted a potential fix in the Gazebo repo: https://bitbucket.org/osrf/gazebo/pull-requests/3199

jslee02, 2020-04-04:
I believe this is resolved by https://bitbucket.org/osrf/gazebo/pull-requests/3199. Feel free to reopen if the problem is still presistent.

closed 2020-04-04 by jslee02

peci1, 2020-04-04:
Thanks for the fix on Gazebo side. I think updating the changelog here still remains :)

jslee02, 2020-04-05:
> Thanks for the fix on Gazebo side.

Glad it worked. Thank you for confirming that it's resolved!

> I think updating the changelog here still remains

Done! https://github.com/dartsim/dart/pull/1444

PR #1444 (2020-04-05): Update changelog for DART 6.9: switched to C++14
Resolves #1423