dartsim/dart #1428: dartsim appears to require outdated dependency headers (assimp 4.1.0 instead of 5.0.0)

GlenHenshaw, 2019-11-26:
> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [ x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x ] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: 6.9.2_1
* OS name and version name(or number): macOS 10.14.6
* Compiler name and version number: Clang 11.0.0

Including darts as a dependency via make should only require paths related to current dependencies of dartsim.

#### Current Behavior
> What is the current behavior?

including dartsim as a dependency in a make recipe fails with the error:

``` Imported target "dart" includes non-existent path

    "/usr/local/Cellar/assimp/4.1.0/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

However, assimp 4.1.0 is not a dependency of dartsim 6.9.2. Assimp 5.0 is. Manually creating the required path without putting any header or library files in it allows the build to proceed, but attempts to use downstream libraries sporadically fail, complaining that libassimp4.1.0.dylib can't be found.

jslee02, 2019-11-26:
How can I reproduce this error? It'd be very helpful if you could provide `CMakeLists.txt` that reproduces the error.

GlenHenshaw, 2019-11-26:
This may be operator error on my part, in that if I manually remove all of the cmake-generated files and then reissue cmake, the problem *appears* to go away. However, it's been intermittent for me for two days now so I'm not convinced yet. Nevertheless, CMakeLists.txt attached.

[CMakeLists.txt](https://github.com/dartsim/dart/files/3894108/CMakeLists.txt)

[CMakeLists.txt](https://github.com/dartsim/dart/files/3894106/CMakeLists.txt)

mxgrey, 2019-11-27:
If you had built against an older version of DART that used assimp 4 and then switched to a newer DART that uses assimp 5 (or, equivalently switched from an older version of Gazebo to a newer version of Gazebo), then it's almost certainly a cmake cache issue.

If you were always building against the same versions of DART and/or Gazebo from the very beginning, then the issue is a bit more mysterious.

GlenHenshaw, 2019-11-27:
I suspect it’s the former at this point, although if the issue recurs I will certainly add more detail here.

It would be very nice if cmake had a “cmake clean”.

traversaro, 2019-11-27:
I guess it is a problem of a Homebrew bottle that needs to be re-generated after Homebrew update from Assimp 4 to Assimp 5 .

traversaro, 2019-11-27:
Unfortunately, it appears that the assimp5 formula in homebrew has broken CMake config files, see: 
* https://github.com/robotology/idyntree/issues/599
* https://github.com/Homebrew/homebrew-core/issues/47210
* https://github.com/assimp/assimp/issues/2760
* https://github.com/assimp/assimp/pull/2765

jslee02, 2020-04-27:
Looking into https://github.com/Homebrew/homebrew-core/issues/47210, this is resolved by https://github.com/assimp/assimp/pull/2765 and https://github.com/Homebrew/homebrew-core/pull/47792. Closing this issue. Feel free to reopen if it's not the case.

closed 2020-04-27 by jslee02