dartsim/dart #1429: CMake Error in FindBullet

kunaltyagi, 2019-11-29:
> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: master and 1.9.2
* OS name and version name(or number): Arch
* Compiler name and version number: GCC 9

#### Expected Behavior
CMake config succeeds

#### Current Behavior
Findings in gazebosim source code:
* `find_package(DART ${GAZEBO_DART_MIN_REQUIRED_VERSION} REQUIRED CONFIG)`
* No custom `FindDART.cmake`

However, the `find_package(DART)` results in the following error:
```
CMake Error at catkin_ws/install/share/dart/cmake/DARTFindBullet.cmake:15 (if):
  if given arguments:

    "WIN32" "AND" "optimized" "IN_LIST" "BULLET_LIBRARIES" "AND" "debug" "IN_LIST" "BULLET_LIBRARIES"

  Unknown arguments specified
Call Stack (most recent call first):
catkin_ws/install/share/dart/cmake/dart_collision-bulletComponent.cmake:6 (include)
catkin_ws/install/share/dart/cmake/DARTConfig.cmake:63 (include)
catkin_ws/install/share/dart/cmake/DARTConfig.cmake:123 (dart_traverse_components)
catkin_ws/install/share/dart/cmake/DARTConfig.cmake:171 (dart_package_init)
  cmake/SearchForStuff.cmake:165 (find_package)
  CMakeLists.txt:147 (include)
```

This happens for master, **NOT 1.9.2**. The change is clearly introduced in 6b023c2994294d6eb1b95e2242e39ba7e9f02427.

#### Steps to Reproduce
Steps for reproducing the issue.
0. Install DART
1. Clone Gazebo (head) and compile
4. Failure!!!

I'm sorry I can't create a smaller CMakeLists.txt only Min Rep Example

costashatz, 2020-02-18:
Any plans on fixing that? Thanks a lot!

jslee02, 2020-02-19:
Let me take a look at this today.

kunaltyagi, 2020-02-19:
> Let me take a look at this today

I've encountered this when some variables are empty and thus string expanded to nothing. CMake considers it an error if "AND", "OR" etc. terms have only rhs/lhs or come after one another due to empty strings.

jslee02, 2020-02-19:
It seems the error happens when CMake failed to parse `IN_LIST`. This happens when `DARTFindBullet.cmake` is loaded by a `CMakeLists.txt` with the minimum required CMake version set to lower than 3.3 (using `cmake_minimum_required()` at the project `CMakeLists.txt`) or when the it's not specified. I believe this should be fixed by adding `cmake_minimum_required()` to `DARTConfig.cmake.in` because it loades `DARTFindBullet.cmake` (and other find modules).

@kunaltyagi Could you please test #1434 if it fixes this issue? Sorry, I have no Arch machine.

kunaltyagi, 2020-02-19:
```
$ cmake --version
cmake version 3.16.4
```

It didn't fix the issue. I compiled using dartsim (HEAD=pr/1434) + Gazebo HEAD

costashatz, 2020-02-19:
> It didn't fix the issue. I compiled using dartsim (HEAD=pr/1434) + Gazebo HEAD

Did you remember to clean the build folder? Sometimes cmake gets stuck..

kunaltyagi, 2020-02-19:
Yes. I cleaned the build directory (`rm -fr *`)

jslee02, 2020-02-19:
> You mentioned that min required version of cmake is 3.5, while I'm on a much more recent version (when I reported the error, I was already using 3.14 or later)

I meant specifying the minimum required CMake version inside the main `CMakeLists.txt` of the project (i.e., gazebo). I believe [Gazebo specifies it](https://bitbucket.org/osrf/gazebo/src/c1537db2c8550db2566a634fcee382a7ab6d7fbe/CMakeLists.txt#lines-1).

It's a bit tricky to test an issue when I'm not able to reproduce it. Do you know if this happens only on Arch or other linux distributions?

kunaltyagi, 2020-02-19:
> It's a bit tricky to test an issue when I'm not able to reproduce it

One of the variables is empty, most likely Win32. This might happen on all non Win32 platforms or if there's no bullet libraries (don't know if that's possible).

> Do you know if this happens only on Arch or other linux distributions?

Sorry, I didn't need to compile from source on other platforms. I had some issues on arch for which I had to compile from source.

> I meant specifying the minimum required CMake version inside the main

I was referring to the diff of the PR, which I think is unrelated to this issue.

jslee02, 2020-02-19:
> One of the variables is empty, most likely Win32. This might happen on all non Win32 platforms or if there's no bullet libraries (don't know if that's possible).

Hm, I believe [`WIN32` is defined to false on other OSs than Windows](https://cmake.org/cmake/help/v3.0/variable/WIN32.html). Also, `IN_LIST` seems to work with an empty list from my tests as long as CMake is able to interpret `IN_LIST` operator.

kunaltyagi, 2020-02-19:
> I believe WIN32 is defined to false on other OSs than Windows

No. It's the negative of your statement. WIN32 is defined to be true on Windows but has an undefined value (might be false, might be literally undefined)

costashatz, 2020-02-20:
@jslee02 if #1434 added lines are like this (i.e., remove `if(NOT CMAKE_MINIMUM_REQUIRED_VERSION)`):

```cmake
if(MSVC)
  cmake_minimum_required(VERSION 3.8.0)
else()
  cmake_minimum_required(VERSION 3.5.1)
endif()
```

Then it works for me..

jslee02, 2020-02-20:
@costashatz What do you see if you try following?

```cmake
if(NOT CMAKE_MINIMUM_REQUIRED_VERSION)
  if(MSVC)
    cmake_minimum_required(VERSION 3.8.0)
  else()
    cmake_minimum_required(VERSION 3.5.1)
  endif()
else()
  message(STATUS "[DEBUG] CMAKE_MINIMUM_REQUIRED_VERSION: ${CMAKE_MINIMUM_REQUIRED_VERSION}")
  if(POLICY CMP0057)
    cmake_policy(SET CMP0057 NEW)
  endif()
  set(foo "apple orange")
  if("apple" IN_LIST foo)
    message(STATUS "[DEBUG] IN_LIST is working!")
  endif()
  set(bar)
  if(NOT "apple" IN_LIST bar)
    message(STATUS "[DEBUG] IN_LIST is working even with an empty list")
  endif()
endif()
```

costashatz, 2020-02-21:
> @costashatz What do you see if you try following?

@jslee02 I see the following:

```bash
-- [DEBUG] CMAKE_MINIMUM_REQUIRED_VERSION: 2.8.6
-- [DEBUG] IN_LIST is working even with an empty list
```

And the configuration works..

jslee02, 2020-02-22:
@costashatz Thank you for the testing for me. It's because of the old CMake version setting, which doesn't support `IN_LIST` operator by default. #1434 should work now.

jslee02, 2020-02-24:
I believe this issue should be resolved by #1434. Please reopen if the problem is still persistent.

closed 2020-02-24 by jslee02

costashatz, 2020-02-24:
> I believe this issue should be resolved by #1434. Please reopen if the problem is still persistent.

Thanks for the fix! Now the cmake configuration is processed correctly. However, when building the latest gazebo version (`10.1.0`) with the DART master I get the following error:

```bash
/usr/local/include/dart/common/detail/ProxyAspect.hpp:141:17: error: ‘make_unique’ is not a member of ‘std’
  141 |     return std::make_unique<ProxyPropertiesAspect>();
```

And a lot of similar errors. I guess the C++14 flag is not passed through the cmake. Any ideas? Thanks a lot!

jslee02, 2020-02-25:
Thank you for the test!

> I guess the C++14 flag is not passed through the cmake.

I think that's the root cause, and this is a bit discussed in https://github.com/dartsim/dart/issues/1423 as well. Let me fix this by letting `DARTConfig.cmake` to pass the C++14 flag. Probably this would require some modification in Gazebo to include the DART flags (e.g., `DART_DEFINITIONS`).

costashatz, 2020-02-25:
> I think that's the root cause, and this is a bit discussed in #1423 as well. Let me fix this by letting `DARTConfig.cmake` to pass the C++14 flag. Probably this would require some modification in Gazebo to include the DART flags (e.g., `DART_DEFINITIONS`).

Thanks! Let me know when you have the fix ready so that I can test..

jslee02, 2020-02-25:
@costashatz Hm, I couldn't reproduce the build error you get. Which OS, CMake, and compiler are you using?

costashatz, 2020-02-28:
> Which OS, CMake, and compiler are you using?

I am using archlinux, cmake 3.16.4, and gcc 9.2.1.. I have built and installed the DART master and trying to install Gazebo through [AUR](https://aur.archlinux.org/packages/gazebo).

jslee02, 2020-03-01:
Could you provide Dockerfile that reproduces the error? I spend a few hours but couldn't get there as I'm not familiar with the distro.

jslee02, 2020-03-01:
Never mind, I think I figured out what's the cause. Please see [this comment](https://github.com/dartsim/dart/issues/1423#issuecomment-593137658).

costashatz, 2020-03-01:
> Never mind, I think I figured out what's the cause. Please see [this comment](https://github.com/dartsim/dart/issues/1423#issuecomment-593137658).

@jslee02 thanks! I will add my comments there then..

PR #1434 (2020-02-24): Add support new if() IN_LIST operator in DARTConfig.cmake
Fixes #1429 

***

**Before creating a pull request**

- [N/A] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [N/A] Add unit test(s) for this change