dartsim/dart #1431: BodyNode::createShapeNodeWith() access violation on Windows 10

koonyook, 2020-01-19:
I previously used DART on Ubuntu and it works fine with this C++ piece of code I have. However, I need to bring some of my code to Windows 10. The compilation of DART and my project on VS2019 went well. Unfortunately, a line of code produces an access violation error during runtime.

This is the independent code section that causes the problem
```
dart::dynamics::SkeletonPtr skel = dart::dynamics::Skeleton::create("abc");
dart::dynamics::BodyNode *parent = nullptr;
dart::dynamics::Joint::Properties *joint_properties = MASS::MakeFreeJointProperties("def", Eigen::Isometry3d::Identity(), Eigen::Isometry3d::Identity());
dart::dynamics::FreeJoint::Properties *prop = dynamic_cast<dart::dynamics::FreeJoint::Properties*>(joint_properties);
dart::dynamics::BodyNode *bn = skel->createJointAndBodyNodePair<dart::dynamics::FreeJoint>(
	parent, (*prop), dart::dynamics::BodyNode::AspectProperties(joint_properties->mName)
	).second;
Eigen::Vector3d size;
size << 1, 1, 1;
dart::dynamics::ShapePtr shape = std::shared_ptr<dart::dynamics::BoxShape>(new dart::dynamics::BoxShape(size));;
bn->createShapeNodeWith<dart::dynamics::VisualAspect>(shape);  //access violation here
```
This is the error log.
`Exception thrown at 0x00007FFBA46F7319 (vcruntime140d.dll) in ConsoleApplication1.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.`

By tracing, I've found that the error occurs in FixedFrame.cpp in the createAspect()
```
FixedFrame::FixedFrame(Frame* refFrame,
                       const Eigen::Isometry3d& relativeTransform)
  : Entity(refFrame, false),
    Frame(refFrame)
{
  createAspect<Aspect>(AspectProperties(relativeTransform)); // runtime error at this line
}
```
And the last tracable line is in Aspect.hpp during the dynamic_cast of function setComposite()
```
mComposite = dynamic_cast<CompositeType*>(newComposite);
```
At this line, The FixedFrame object inside the newComposite seems to point to nowhere as the debugger is unable to read the memory inside that object.

As it is related to dynamic_cast, Run-Time Type Information (/GR) has been enabled but the problem still exist. 

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [Y] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [Y] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: 6.9.2 (stable release) with all the library from vcpkg (according to http://dartsim.github.io/install_dart_on_windows.html)
* OS name and version name(or number): Windows 10 (1903)
* Compiler name and version number: Visual Studio 2019 (default compiler)

ryo-y-murakami, 2020-05-02:
It works fine with /vd2 compiler option.

donnyward, 2020-12-11:
I have the same problem crash on `createShapeNodeWith`, last traceable line is where the `dynamic_cast` occurs. This is using the dart package from `vcpkg` with a tiny sample program building a skeleton with one joint and body.

Adding `/vd2` to the project does not make a difference.

traversaro, 2020-12-11:
Related issue (I guess): https://github.com/dartsim/dart/issues/1522 .

erwincoumans, 2021-03-13:
Thanks to @ryo-y-murakami, I confirm that adding /vd2 in the root CMakeLists.txt makes the Windows build work, Visual Studio 2019.

jslee02, 2021-04-14:
I believe this issue is fixed by https://github.com/dartsim/dart/pull/1541

closed 2021-04-14 by jslee02