dartsim/dart #1433: Acceleration grows to infinity for a rolling ball

azeey, 2020-02-15:
### Bug Report
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
* DART version:  master, 6.9, 6.6
* OS name and version name(or number): Ubuntu 18.04
* Compiler name and version number: GCC 8.3.0

#### Expected Behavior
A ball rolling without slip due to a constant force should have a constant acceleration.

#### Current Behavior

[image]

The acceleration of the ball grows to infinity ultimately causing DART to crash. This doesn't occur if the ball slides without rolling.  The behavior of the simulation changes depending on the collision detector used. In the video, I'm using ODE as the collision detector, but the same thing happens with FCL with primitive shapes. With bullet, the ball stops rolling after a little while and the print out shows what appears to be random accelerations in all three axes.

* Related issue: https://bitbucket.org/ignitionrobotics/ign-physics/issues/7/ball-rolling-down-an-incline-achieves-warp

#### Steps to Reproduce / Code to Reproduce
I have created an example to demonstrate the problem [here](https://github.com/azeey/sandbox/tree/master/dart_warp_speed). A ball is placed on a horizontal box with gravity set to `(5, 0, -9.8)`. The expected result is a rolling motion with a constant acceleration of 3.57 m/s^2. I'm using a horizontal component of gravity, but the same experiment can be done with a ball rolling on an incline.
See https://bitbucket.org/ignitionrobotics/ign-physics/issues/7/ball-rolling-down-an-incline-achieves-warp

scpeters, 2020-02-19:
this may be related to #229 and #1193 (and #424)

jslee02, 2020-02-19:
> this may be related to #229 and #1193 (and #424)

I guess so too, and probably #1193 would have most significant affects on the behavior.

> The behavior of the simulation changes depending on the collision detector used.

This makes sense because different collision detector returns different results. Especially, DART doesn't use primitive shape support of FCL since it has some unresolved issues so mesh collsion detection of coarse sphere-shaed mesh would be less realistic than others.

Let me investigate on this further when I have a chance. Thanks for the detailed report and the reproduction code!

jslee02, 2021-12-18:
@azeey Do you think this issue is resolved by https://github.com/dartsim/dart/pull/1437?

jslee02, 2021-12-19:
Closing this issue per #1437. Feel free to reopen if it's not the case.

closed 2021-12-19 by jslee02

azeey, 2021-12-19:
> @azeey Do you think this issue is resolved by #1437?

Yes, it does!