dartsim/dart #1435: build problem.  make[3]: *** [python/dartpy/dartpy.so] Error 1

BAn21, 2020-02-26:
#### Environment
> Select the following information.
* DART version: DART 6.9.0
* OS name and version name(or number): [e.g., Ubuntu 16.04
* Compiler name and version number: pip3 verion 2.20 / C++ version 3.11 

#### Expected Behavior
i follow up this page, so 
https://dartsim.github.io/install_dartpy_on_ubuntu.html

$ cmake .. -DDART_BUILD_DARTPY=ON -DCMAKE_INSTALL_PREFIX=/usr/ -DCMAKE_BUILD_TYPE=Release
$ make -j4 dartpy

or 

$ cmake .. -DDART_BUILD_DARTPY=ON -DCMAKE_INSTALL_PREFIX=/usr/ -DCMAKE_BUILD_TYPE=Release
$ sudo make install

$ cmake .. -DDART_BUILD_DARTPY=ON 
 .
.

all same error

.
.
.
.
[100%] Building CXX object python/dartpy/CMakeFiles/dartpy.dir/simulation/module.cpp.o
[100%] Building CXX object python/dartpy/CMakeFiles/dartpy.dir/utils/DartLoader.cpp.o
[100%] Building CXX object python/dartpy/CMakeFiles/dartpy.dir/utils/SkelParser.cpp.o
[100%] Building CXX object python/dartpy/CMakeFiles/dartpy.dir/utils/module.cpp.o
[100%] Linking CXX shared module dartpy.so
/usr/local/include/pybind11/cast.h:866:56: note: type ‘struct type_caster’ itself violates the C++ One Definition Rule
 template <typename type, typename SFINAE = void> class type_caster : public type_caster_base<type> { };
                                                        ^
/usr/local/include/pybind11/eigen.h:249:8: note: the incompatible type is defined here
 struct type_caster<Type, enable_if_t<is_eigen_dense_plain<Type>::value>> {
        ^
/usr/local/include/pybind11/cast.h:866:56: note: type ‘struct type_caster’ itself violates the C++ One Definition Rule
 template <typename type, typename SFINAE = void> class type_caster : public type_caster_base<type> { };
                                                        ^
/usr/local/include/pybind11/eigen.h:249:8: note: the incompatible type is defined here
 struct type_caster<Type, enable_if_t<is_eigen_dense_plain<Type>::value>> {
        ^
/usr/local/include/pybind11/cast.h:866:56: note: type ‘struct type_caster’ itself violates the C++ One Definition Rule
 template <typename type, typename SFINAE = void> class type_caster : public type_caster_base<type> { };
                                                        ^
/usr/local/include/pybind11/eigen.h:249:8: note: the incompatible type is defined here
 struct type_caster<Type, enable_if_t<is_eigen_dense_plain<Type>::value>> {
        ^
/usr/include/c++/7/tuple:556:11: note: type ‘struct tuple’ itself violates the C++ One Definition Rule
     class tuple : public _Tuple_impl<0, _Elements...>
           ^
/usr/local/include/pybind11/cast.h:866:56: note: type ‘struct type_caster’ itself violates the C++ One Definition Rule
 template <typename type, typename SFINAE = void> class type_caster : public type_caster_base<type> { };
                                                        ^
/usr/local/include/pybind11/stl.h:179:49: note: the incompatible type is defined here
 template <typename Type, typename Alloc> struct type_caster<std::vector<Type, Alloc>>
                                                 ^
../../lib/libdart.a(EulerJointAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail26EulerJointUniquePropertiesE[_ZTVN4dart8dynamics6detail26EulerJointUniquePropertiesE]+0x10): undefined reference to `dart::dynamics::detail::EulerJointUniqueProperties::~EulerJointUniqueProperties()'
../../lib/libdart.a(EulerJointAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail26EulerJointUniquePropertiesE[_ZTVN4dart8dynamics6detail26EulerJointUniquePropertiesE]+0x18): undefined reference to `dart::dynamics::detail::EulerJointUniqueProperties::~EulerJointUniqueProperties()'
../../lib/libdart.a(PlanarJointAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail27PlanarJointUniquePropertiesE[_ZTVN4dart8dynamics6detail27PlanarJointUniquePropertiesE]+0x10): undefined reference to `dart::dynamics::detail::PlanarJointUniqueProperties::~PlanarJointUniqueProperties()'
../../lib/libdart.a(PlanarJointAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail27PlanarJointUniquePropertiesE[_ZTVN4dart8dynamics6detail27PlanarJointUniquePropertiesE]+0x18): undefined reference to `dart::dynamics::detail::PlanarJointUniqueProperties::~PlanarJointUniqueProperties()'
../../lib/libdart.a(RevoluteJointAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail29RevoluteJointUniquePropertiesE[_ZTVN4dart8dynamics6detail29RevoluteJointUniquePropertiesE]+0x10): undefined reference to `dart::dynamics::detail::RevoluteJointUniqueProperties::~RevoluteJointUniqueProperties()'
../../lib/libdart.a(RevoluteJointAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail29RevoluteJointUniquePropertiesE[_ZTVN4dart8dynamics6detail29RevoluteJointUniquePropertiesE]+0x18): undefined reference to `dart::dynamics::detail::RevoluteJointUniqueProperties::~RevoluteJointUniqueProperties()'
../../lib/libdart.a(ScrewJointAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail26ScrewJointUniquePropertiesE[_ZTVN4dart8dynamics6detail26ScrewJointUniquePropertiesE]+0x10): undefined reference to `dart::dynamics::detail::ScrewJointUniqueProperties::~ScrewJointUniqueProperties()'
../../lib/libdart.a(ScrewJointAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail26ScrewJointUniquePropertiesE[_ZTVN4dart8dynamics6detail26ScrewJointUniquePropertiesE]+0x18): undefined reference to `dart::dynamics::detail::ScrewJointUniqueProperties::~ScrewJointUniqueProperties()'
../../lib/libdart.a(TranslationalJoint2DAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail36TranslationalJoint2DUniquePropertiesE[_ZTVN4dart8dynamics6detail36TranslationalJoint2DUniquePropertiesE]+0x10): undefined reference to `dart::dynamics::detail::TranslationalJoint2DUniqueProperties::~TranslationalJoint2DUniqueProperties()'
../../lib/libdart.a(TranslationalJoint2DAspect.cpp.o):(.data.rel.ro._ZTVN4dart8dynamics6detail36TranslationalJoint2DUniquePropertiesE[_ZTVN4dart8dynamics6detail36TranslationalJoint2DUniquePropertiesE]+0x18): undefined reference to `dart::dynamics::detail::TranslationalJoint2DUniqueProperties::~TranslationalJoint2DUniqueProperties()'
collect2: error: ld returned 1 exit status
python/dartpy/CMakeFiles/dartpy.dir/build.make:1390: recipe for target 'python/dartpy/dartpy.so' failed
make[3]: *** [python/dartpy/dartpy.so] Error 1
CMakeFiles/Makefile2:6423: recipe for target 'python/dartpy/CMakeFiles/dartpy.dir/all' failed
make[2]: *** [python/dartpy/CMakeFiles/dartpy.dir/all] Error 2
CMakeFiles/Makefile2:6435: recipe for target 'python/dartpy/CMakeFiles/dartpy.dir/rule' failed
make[1]: *** [python/dartpy/CMakeFiles/dartpy.dir/rule] Error 2
Makefile:1787: recipe for target 'dartpy' failed
make: *** [dartpy] Error 2

i can't find the same error in here..
how can i fix this? please help me.

BAn21, 2020-02-28:
one more, I did $ catkin build
there are same log, and simillar error, too.
thanks.

CMakeFiles/dart.dir/common/Uri.cpp.o: In function `__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > boost::re_detail::re_is_set_member<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, char, boost::regex_traits<char, boost::cpp_regex_traits<char> >, unsigned int>(__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, __gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, boost::re_detail::re_set_long<unsigned int> const*, boost::re_detail::regex_data<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, bool)':
Uri.cpp:(.text._ZN5boost9re_detail16re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb[_ZN5boost9re_detail16re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb]+0x18c): undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
Uri.cpp:(.text._ZN5boost9re_detail16re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb[_ZN5boost9re_detail16re_is_set_memberIN9__gnu_cxx17__normal_iteratorIPKcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEjEET_SH_SH_PKNS0_11re_set_longIT2_EERKNS0_10regex_dataIT0_T1_EEb]+0x40f): undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
collect2: error: ld returned 1 exit status
make[2]: *** [lib/libdart.so.6.9.2] Error 1
make[1]: *** [dart/CMakeFiles/dart.dir/all] Error 2
make: *** [all] Error 2
cd /home/ban/gazebo9_ws/build/dart; catkin build --get-env dart | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -
...................

jslee02, 2020-03-01:
I was able to build dartpy of master branch on Xenial. Could you confirm that you also can build master branch?

jslee02, 2020-04-27:
Closing this issue due to lack of response. Feel free to reopen if the problem is still persistent.

closed 2020-04-27 by jslee02