dartsim/dart #1438: Interfacing with other code

christophecricket, 2020-03-02:
I'd like to interface a dart simulation with a system built in ROS to set inputs and read outputs. In the tutorials, the iteration of the simulation seems to be handled by a glutMainLoop(), so it is difficult for me to integrate a ROS subscriber and publisher into that because I can't insert code into a loop. Is there a way to make this happen? For the record I should state that I have no idea how glut works at all.

christophecricket, 2020-03-02:
Upon further searching through the examples, I have found something that might work. Could I just overload the timeStepping() method of SimWindow? Would that work?

christophecricket, 2020-03-02:
Closed because I read the pendulum tutorial and it told me to do just that. Sorry

closed 2020-03-02 by christophecricket

mxgrey, 2020-03-03:
If you'd like a headless simulation, that's actually the easiest way to use dartsim. You don't need the `dart::gui` library at all:

1. Instantiate a [`dart::simulation::World`](https://github.com/dartsim/dart/blob/master/dart/simulation/World.hpp), populate it with the models you want (or load a world from [urdf](https://github.com/dartsim/dart/blob/v6.9.2/dart/utils/urdf/DartLoader.hpp#L107) or [sdf](https://github.com/dartsim/dart/blob/v6.9.2/dart/utils/sdf/SdfParser.hpp#L51)). This shouldn't be much different from the GUI examples.
2. Simply call [`world->step()`](https://github.com/dartsim/dart/blob/v6.9.2/dart/simulation/World.hpp#L217) whenever you want the simulation to move forward in time.