dartsim/dart #1439: Illegal instruction: 4 when running atlas_simbicon example on mac

junhyeokahn, 2020-03-12:
> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [v] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [v] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: release-6.9
* OS name and version name(or number): macOS Catalina
* Compiler name and version number: AppleClang 11.0.0.11000033

#### Expected Behavior
> Please describe the behavior you are expecting.
running atlas_simbicon
#### Current Behavior
> What is the current behavior?
> If you have trouble in `cmake` or `make`, it would be very helpful to build DART with verbose option:
> `$ cmake .. -DDART_VERBOSE=On`
> `$ make VERBOSE=1`
> and then dump the results into a [gist](https://gist.github.com/) and share the link to gist here.

I was able to compile repo and examples with ```make -j4 examples```.
However, when I run ```./bin/atlas_simbicon``` it returns ```Illegal instruction: 4``` with the error message below.
```
Process:               atlas_simbicon [24544]
Path:                  /Users/USER/*/atlas_simbicon
Identifier:            atlas_simbicon
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        bash [19570]
Responsible:           iTerm2 [2076]
User ID:               501

Date/Time:             2020-03-12 12:00:41.446 -0500
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Anonymous UUID:        DE30F051-7C7C-2CFD-65F7-0753ADCDD6D1

Sleep/Wake UUID:       5B630AEC-E2D6-490F-AF26-107422779F98

Time Awake Since Boot: 19000 seconds
Time Since Wake:       2100 seconds

System Integrity Protection: disabled

Crashed Thread:        8

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [24544]

Application Specific Information:
-[NSOpenGLContext update] must be called from the main thread if the context has a view.

Thread 0:: Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	0x00007fff678e6e25 getMethodNoSuper_nolock(objc_class*, objc_selector*) + 47
1   libobjc.A.dylib               	0x00007fff678e8b99 _objc_msgSend_uncached + 73
2   com.apple.UIFoundation        	0x00007fff62348444 attributeDictionaryIsEqual + 69
3   com.apple.Foundation          	0x00007fff33d9eb80 hashProbe + 516
4   com.apple.Foundation          	0x00007fff33d9e94f -[NSConcreteHashTable getItem:] + 52
5   com.apple.UIFoundation        	0x00007fff6233ad2f +[NSAttributeDictionary newWithDictionary:] + 115
6   com.apple.Foundation          	0x00007fff33e061e8 -[NSConcreteAttributedString initWithString:attributes:] + 128
7   com.apple.AppKit              	0x00007fff2e96cbfc -[NSTitledFrame _titleDidChange] + 78
8   com.apple.AppKit              	0x00007fff2e96c824 -[NSTitledFrame setTitle:] + 703
[... 421 lines cut ...]
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.77.106.3 AirPortDriverBrcmNIC-1440.1)
Bluetooth: Version 7.0.3f5, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512G, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
USB Device: Hub
USB Device: HHKB Professional
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```
It must be related to opengl or osg since I am able to run other executables such as ```glut_rigid_loop```.

#### Steps to Reproduce
> Please provide detailed steps for reproducing the issue.
1. install dart from source
2. compile examples
3. run atlas_simbicon example

#### Code to Reproduce
> Please remember that with a working code it's easier to reproduce the bug and it's much faster to fix it.
> It would be great to fork this repo and create a regression test using [this template](https://github.com/dartsim/dart/blob/master/unittests/regression/test_Issue000Template.cpp).

jslee02, 2020-03-12:
I was not able to reproduce this issue on macOS Mojave. Maybe OSG is not compatible with the OpenGL binaries in Catalina. Would you be able to check if this issue happens macOS Mojave or older macOS version?

Do you see the same error from other OSG based examples or just `atlas_simbicon`?

junhyeokahn, 2020-03-12:
Thanks for the reply.

I was able to run it on macOS Mojave.
In macOS Catalina, I am seeing this issue for all OSG based examples.

jslee02, 2020-03-13:
Confirmed that this issue is reproducible on my mac (Catalina). I would test if a pure OSG example works on Catalina to see if it's DART or OSG. Will update if I find out anything.

junhyeokahn, 2020-03-13:
Thanks for checking that!

jslee02, 2020-04-05:
@junhyeokahn Sorry for the late response! I'm currently not able to test DART and OSG on Catalina as my old MacBook is recently broken. Would you be able to test if OSG examples in the OSG repository is executable without getting the error you reported? This is to see if the issue is from OSG or DART.

junhyeokahn, 2020-04-14:
Hi,
I have tried to run an OSG example in their git repo following [this](http://openscenegraph.sourceforge.net/documentation/OpenSceneGraph/doc/examples.html) instruction.
However, it is not well documented so I couldn't run any one of these.
Could you recommend me any pointer so I could run examples?

jslee02, 2020-04-27:
> However, I have trouble running the examples by just running the commands on the webpage.

What's the issue you see? Just in case, it seems [this](http://www.openscenegraph.org/index.php/documentation/guides/user-guides/113-running-the-examples) is the newer website. No difference in "running the examples" page though. You might get better advice from the [OSG community](http://www.openscenegraph.org/index.php/support) if you find a problem running OSG examples.

Sorry for not being able to help the issue on Catalina.

junhyeokahn, 2020-06-02:
Sorry for the late reply.
I was able to run the osgexamples with osg version 3.7.0 on Catalina.

It might be related to the osg version Dart uses.

jslee02, 2020-06-03:
Yeah, [the version of OSG installed by Homebrew is 3.6.5](https://github.com/Homebrew/homebrew-core/blob/master/Formula/open-scene-graph.rb#L4). It seems OSG 3.7.0 is still under development (as [it's not tagged](https://github.com/openscenegraph/OpenSceneGraph/releases)). It's possible that the issue is fixed between 3.6.6 and 3.7.0 (master branch).

One thing you could try is reinstalling OSG 3.7.0 either from the source or using homebrew (by running `brew install openscenegraph --HEAD`) and building DART against it.

jslee02, 2020-07-26:
I finally was able to upgrade my MacBook to Catalina and confirmed that the root cause of this issue was https://github.com/openscenegraph/OpenSceneGraph/issues/926. It seems that OpenSceneGraph 3.7.0 or greater should be used on Catalina. #1479 is for enforcing it.

junhyeokahn, 2020-07-27:
@jslee02  That's awesome! Thanks for the fix :)

PR #1479 (2020-07-28): Enforce to use OSG 3.7.0 or greater on macOS Catalina
OpenSceneGraph 3.6.5 or less are not compatible with macOS 10.15 (Catalina) or greater (https://github.com/openscenegraph/OpenSceneGraph/issues/926, https://github.com/dartsim/dart/issues/1439). This PR increases the required minimum OSG version for macOS Catalina to 3.7.0 that includes the [fix](https://github.com/openscenegraph/OpenSceneGraph/pull/944).

Resolves #1439

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x]  [... cut]
closed 2020-07-28 by jslee02