dartsim/dart #1449: ikFast class checks joint limits incorrectly

sniyaz, 2020-04-24:
#### Environment

Irrelevant, but on Dart 6.9 with GCC 7.5 on Bionic.

#### Expected Behavior

All valid solutions being returned through ikFast.

#### Current Behavior
https://github.com/dartsim/dart/blob/660b0f4d84b56cfb723eb5e50e2e465f49930f5a/dart/dynamics/IkFast.cpp#L139

It doesn't look like ikFast gives any promises for the range of returned ik solutions. For example, with a joint with limits at [0, 2pi], the ikFast binary may return a solution at -pi. 

Checking this raw value against the skeleton limits (as is done now) appears to lead to valid solutions being thrown away (in the example above, we can just wrap the joint value to pi, which is actually in-limits).

It looks like some wrapping is needed in the joint limit checking here, since ikFast doesn't provide a guarantee on the range as mentioned earlier.

#### Steps to Reproduce
I have a Jaco ikFast binary I'd be happy to share where this happens.

#### Code to Reproduce
Simply run the provided binary: almost all solutions are thrown away.

jslee02, 2020-04-25:
Thank you for the report. The workaround makes sense to me. I wonder why IKFast wraps the solution in the first place though. :thinking: 

Maybe we can wrap the joint values only for SO(2) joint. I'm not sure what the representation is used for SO(3) joint in IKFast.

> I have a Jaco ikFast binary I'd be happy to share where this happens.

Could you share the binary and the URDF as well? It would be very helpful to ensure if the issue is solved by having it as a regression test with the models.

jslee02, 2020-04-26:
A possible solution for this issue: https://github.com/dartsim/dart/pull/1452

jslee02, 2020-04-27:
@sniyaz Please let me know when you confirm that #1452 fixes the issue.

sniyaz, 2020-04-28:
Thank you @jslee02! I won't have access to my lab machine for a bit, but I'll let you know ASAP whether this resolves things.

Would you still like the URDF and the binary?

jslee02, 2020-05-01:
> I won't have access to my lab machine for a bit, but I'll let you know ASAP whether this resolves things.

Thanks!

> Would you still like the URDF and the binary?

Optional, but it'd be great if you can share!

jslee02, 2020-08-28:
Closing this issue as the fix was verified by the unit test. @sniyaz Please let me know if the issue is persistent even with the fix. :)

PR #1452 (2020-04-27): Wrap IKFast solutions for cyclic joint positions
IKFast sometimes returns solutions that are not wrapped within the range of the joint limits, which is regarded as an invalid solution in DART.

This PR fixes the issue by adding logic that tries to wrap IKFast solutions for rotational joints. The idea is to wrap the solution within joint limits. If there is more than one solution (when the joint limit range is wider than 2pi), it returns the closest solution to the current joint position to minimize joint motion.

The logic should be applied to any DegreeOfFreedom whose configuration space is S [... cut]
closed 2020-08-28 by jslee02