dartsim/dart #1450: Incorrect damping force in inverse dynamics

zhengpingzhou, 2020-04-25:
### Bug Report
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
* DART version: [e.g., master, 6.3.0]
* OS name and version name(or number): Ubuntu 18.04
* Compiler name and version number: g++ 7.5.0

#### Expected Behavior
* Inputs: initial position `q0`, velocity `v0`, force `f`
* Outputs: 
    ```
    a_FD = forwardDynamics(q0, v0, f)         # ground truth acceleration
    f_ID = inverseDynamics(q0, v0, a_FD))     # expect: f == f_ID
    a_ID =  forwardDynamics(q0, v0, f_ID)     # expect: a == a_ID
    ```

#### Current Behavior
`f != f_ID, a != a_ID`
In fact, as far as I have tested, `f == f_ID + kDamping * a * timestep`. Details are in the attached code.
   
#### Code to Reproduce
[dart-inverse-dynamics-bug-report.cpp](https://gist.github.com/zhengpingzhou/f809ec6580793a1afb796a7df598e5b2) 

#### Steps to Reproduce
1. Build and run the code (no command line args required). The code produces wrong results.
2. Uncomment line 158-160. The code produces correct results.

jslee02, 2020-04-25:
Thank you for the bug report! Your code was very helpful to quickly reproduce and fix the issue, thanks.

#1451 should resolve the issue. Let me know if you also confirm that it's the right fix.

zhengpingzhou, 2020-04-25:
Thank you for the timely response. I will get back to you once I run your PR on my side.

I'm actually a bit confused why this bug happens -- it seems that it is due to inconsistent handling of damping in FD and ID. Concretely, the bug could be explained if FD is using the damping after time stepping, and ID using the damping before time stepping.

However, I'm not able to find such inconsistency. My proposed workaround was discovered by trial-and-error. Could you please let me know how it could be explained theoretically?

jslee02, 2020-04-25:
> Concretely, the bug could be explained if FD uses the velocity **after** time stepping to compute the damping force, and ID computes the damping force with the velocity **before** time stepping.

This actually explains the bug correctly. Before the fix, only FD used the velocity after time stepping. Suppose we have the equations of motion as follows:
```
M(q(k))*ddq(k) + Cg(q(k), dq(k)) = f(k) - Kd*dq(k+1)
```
where `dq(k+1)` is the velocity after time stepping. By the explicit Euler method, `dq(k+1)` becomes `Kd*( dq(k) + h*ddq(k) )`. Plugging it into the equation and solving the equation for `ddq(k)`, we get:
```
ddq(k) = [M(q(k)) + Kd*ddq(k)]^{-1} * [f(k) - Kd*dq(k) - Cg(q(k), dq(k))]
```
Note that **`Kd*ddq(k)`** is added to the inertia matrix due to the implicit damping force, as we used the velocity after time stepping for damping. In DART code, this line is where this term is applied:

https://github.com/dartsim/dart/blob/9cfc3f651c47309404b21da9e703e2657e55e56a/dart/dynamics/BodyNode.cpp#L2285

#1451 fixes this inconsistency by applying the implicit damping force (and spring force) for ID as well.

Hope this answers your question.

zhengpingzhou, 2020-04-25:
Thanks for the great explanation! This answers my question. 

---
> ```
> ddq(k) = [M(q(k)) + Kd*ddq(k)]^{-1} * [f(k) - Kd*dq(k) - Cg(q(k), dq(k))]
> ```
Should it actually be the following?
```
ddq(k) = [M(q(k)) + Kd*h]^{-1} * [f(k) - Kd*dq(k) - Cg(q(k), dq(k))]
```
I also found this line when debugging and believe it is relevant to your explanation.
https://github.com/dartsim/dart/blob/18bff8ef2ff2c7ac688ea42e3e3a79a997e413e8/dart/dynamics/detail/GenericJoint.hpp#L1933

jslee02, 2020-04-25:
> Should it actually be the following?

Right, thanks for the correction! 

> I also found this line when debugging and believe it is relevant to your explanation

Yes, that line is more relevant. The code in BodyNode (line: 2285) is for computing the inertia matrix explicitly while the code in GenericJoint (line: 1933) is the part of the forward dynamics, which is O(n). Note that we don't need to explicitly compute the mass matrix for the forward dynamics, which is O(n^2).

Let me know if #1451 actually fixes the issue!

zhengpingzhou, 2020-04-26:
Thanks for the clarification! I tested your PR on my test case and it fixes the issue. 

Wish you all the best!

PR #1451 (2020-04-27): Fix inverse dynamics with damping and stiffness
This PR fixes incorrect implicit joint damping/spring forces in inverse dynamics computation.

Fixes #1450 

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change
closed 2020-04-27 by jslee02