dartsim/dart #1469: libdart6-dev wrong config.hpp installation

costashatz, 2020-05-16:
### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: 6.9.2
* OS name and version name(or number): any Ubuntu distro
* Compiler name and version number: Not compiling, I am using the PPA for dartsim

#### Expected Behavior

When installing `libdart6-dev` I would expect a `dart/config.hpp` to be installed where the values `HAVE_BULLET` and `HAVE_ODE` would have the values equal to zero.
#### Current Behavior
On the contrary, the values are set to 1.

#### Steps to Reproduce
1. On a clean Ubuntu setup
2. `sudo apt-add-repository ppa:dartsim/ppa`
3. `sudo apt-get install libdart6-dev`
4. `cat /usr/include/dart/config.hpp`

You should be able to see that the values mentioned are set to 1. Weirdly `libdart6-dev` installs `libdart6-collision-bullet-dev` as a dependency, but not `libdart6-collision-ode-dev`. I am not sure what would be the fix, but at the moment if someone uses the `config.hpp` to identify whether dart has been built with `bullet` or `ode` support, will be misleadingly thinking that they are when `ode` is not even installed in the system. I see two solutions:

- `libdart6-dev` should install a `config.hpp` with all these values to 0 (apart from `octomap` which should be listed as a dependency). When another component is being installed (e.g., `libdart6-collision-bullet-dev`), the `config.hpp` should be updated. Is this possible with debian packages?
- If the above is not possible, then I would propose to make everything that could be in the configuration file a dependency or provide a different way of checking whether something is inside the library or not.

jslee02, 2020-05-18:
> * `libdart6-dev` should install a `config.hpp` with all these values to 0 (apart from `octomap` which should be listed as a dependency). When another component is being installed (e.g., `libdart6-collision-bullet-dev`), the `config.hpp` should be updated. Is this possible with debian packages?

This seems not supported. `apt` complains if a package attempts to overwrite an existing file (to update the file).

> * If the above is not possible, then I would propose to make everything that could be in the configuration file a dependency or provide a different way of checking whether something is inside the library or not.

I don't have a good idea of a way of checking whether an optional component is installed or not when the component is a separate package.

Alternatively, we could remove those definitions from `config.hpp` completely. Those definitions wouldn't make sense unless DART is a single debian package, which is not the case here. Looking at the code we only use the definitions for unit testing. In this case, we could add the definitions as privately to the unit test targets.

costashatz, 2020-05-18:
> This seems not supported. `apt` complains if a package attempts to overwrite an existing file (to update the file).

I suspected it to be like this.

> Alternatively, we could remove those definitions from `config.hpp` completely.

No. I think that these are good to have..

> Looking at the code we only use the definitions for unit testing.

Yes, but when checking for DART, how can someone identify with which libraries DART is compiled with? If you remove those from `config.hpp`, then you have to make sure that each individual library of DART is not linking/including any other library of the possible dependencies so that one can use the public API/headers without their code breaking. This is not the case at the moment. For example, check the [fcl collision detector](https://github.com/dartsim/dart/blob/master/dart/collision/fcl/BackwardCompatibility.hpp#L83); btw, you are using one of these defines here (which is a file that everybody is including!).

> Weirdly `libdart6-dev` installs `libdart6-collision-bullet-dev` as a dependency, but not `libdart6-collision-ode-dev`

Why is this happening?

jslee02, 2020-05-19:
> Yes, but when checking for DART, how can someone identify with which libraries DART is compiled with?

You can specify which optional component should be linked to something like:
```cmake
find_pacakge(DART 6.9.2 REQUIRED
  COMPONENTS
    collision-bullet # libdart6-collision-bullet-dev should be installed
    collision-ode    # libdart6-collision-ode-dev should be installed
  CONFIG)
```

`config.hpp` will always set the variables to true because when the DART packages are built in the PPA build server all the components are built with all the dependencies. We just install the build artifacts separately as each subpackage specify which build artifacts to be installed.

> Why is this happening?

I'm not sure. `libdart6-dev` doesn't have a dependency either `libdart6-collision-ode-dev` or `libdart6-collision-bullet-dev`.

costashatz, 2020-05-19:
> You can specify which optional component should be linked to something like

This I already know. This is not what I meant.

> We just install the build artifacts separately as each subpackage specify which build artifacts to be installed.

Yes I understand this, but as I said in my previous comment, this requires that each subcomponent is **completely** independent. I showed you one example that this is not the case. To make it more practical, if someone installs `libdart6-dev` and does not have octomap installed, trying to use DART (which includes FCL collision detector headers) will result in include errors as it will try to load octomap.

In any case, let's leave it here as this is not super important and it will probably confuse more users if we change the `config.hpp`.

jslee02, 2020-05-22:
> Yes I understand this, but as I said in my previous comment, this requires that each subcomponent is completely independent.

The idea of the optional components is to allow using DART with and without the components. If it's not the case it should be fixed.

> To make it more practical, if someone installs `libdart6-dev` and does not have octomap installed, trying to use DART (which includes FCL collision detector headers) will result in include errors as it will try to load octomap.

This shouldn't be a problem because [`octomap` is installed when `libdart6-dev` is built and installed](https://github.com/dartsim/dart-debian/blob/823817b4f684c860eff52a7ba3c7cffe30757e5b/focal/debian/control#L160).

jslee02, 2023-01-07:
Closing as no action items are identified

closed 2023-01-07 by jslee02