dartsim/dart #1472: Tutorials don't build

MikePelton, 2020-06-01:
Hi all - am on Ubuntu 20.04, trying to get DART running. I've installed 6.9.2 from the ppa but while the install seems to be absolutely fine neither the tutorials nor examples will build - there are many compilation errors despite a clean cmake. Please confirm which versions of gcc, eigen and the dependencies this most recent stable build has been _tested_ against.

Some of the dozens of errors in main.cpp for hello_world.cpp (!)  are listed below:

Executing command: /usr/bin/cmake --build /home/mike/Development/doc/dart/examples/hello_world/build --config Debug --target all -- -j 26
[build] Scanning dependencies of target hello_world
[build] [ 50%] Building CXX object CMakeFiles/hello_world.dir/main.cpp.o
[build] In file included from /usr/include/dart/dynamics/HeightmapShape.hpp:190,
[build]                  from /usr/include/dart/gui/osg/render/HeightmapShapeNode.hpp:46,
[build]                  from /usr/include/dart/gui/osg/render/render.hpp:8,
[build]                  from /usr/include/dart/gui/osg/osg.hpp:18,
[build]                  from /home/mike/Development/doc/dart/examples/hello_world/main.cpp:34:
[build] /usr/include/dart/dynamics/detail/HeightmapShape-impl.hpp: In constructor ‘dart::dynamics::HeightmapShape<S_>::HeightmapShape()’:
[build] /usr/include/dart/dynamics/detail/HeightmapShape-impl.hpp:46:45: error: ‘HEIGHTMAP’ was not declared in this scope
[build]    46 | HeightmapShape<S>::HeightmapShape() : Shape(HEIGHTMAP), mScale(1, 1, 1)

============================

In file included from /usr/include/dart/gui/osg/render/PointCloudShapeNode.hpp:39,
[build]                  from /usr/include/dart/gui/osg/render/render.hpp:13,
[build]                  from /usr/include/dart/gui/osg/osg.hpp:18,
[build]                  from /home/mike/Development/doc/dart/examples/hello_world/main.cpp:34:
[build] /usr/include/dart/dynamics/PointCloudShape.hpp: At global scope:
[build] /usr/include/dart/dynamics/PointCloudShape.hpp:157:8: error: ‘void dart::dynamics::PointCloudShape::updateVolume() const’ marked ‘override’, but does not override

============================

In file included from /usr/include/dart/dynamics/HeightmapShape.hpp:190,
[build]                  from /usr/include/dart/gui/osg/render/HeightmapShapeNode.hpp:46,
[build]                  from /usr/include/dart/gui/osg/render/render.hpp:8,
[build]                  from /usr/include/dart/gui/osg/osg.hpp:18,
[build]                  from /home/mike/Development/doc/dart/examples/hello_world/main.cpp:34:
[build] /usr/include/dart/dynamics/detail/HeightmapShape-impl.hpp:218:3: error: cannot convert ‘const dart::math::BoundingBox*’ to ‘dart::math::BoundingBox*’

MikePelton, 2020-06-02:
Building 6.9.2 on Windows, what's going wrong here is much clearer - both the examples and the tutorials are stuck at version 6.6 - they mandate it in their cmake files, and bodging that to get the code build-able, there are enough things that changed between 6.6 and 6.9.2 to mean that not even hello_world will compile. It would be good for the project if someone who understands the subtleties of what changed after 6.6 could update everything to match.

MikePelton, 2020-06-04:
Hi Jeongseok and thanks for coming back. Can't change my config now to check but am pretty sure my first attempt was to use the ppa - am currently using 6.6 built from source against boost 1.67 (not my choice!) on Ubuntu 20.04 and the examples and tutorials for that work beautifully once I'd built fcl from source (needed because of the boost dependency). I'll check the ppa version again later but if someone has a clean system to hand I'd be delighted to hear my issues with 6.9.2 are my fault and not implicit in the release.

jslee02, 2025-11-03:
We now standardize on pixi for environment setup and builds (pixi run config, pixi run build, pixi run test, etc.). That covers the workflow requested here, so I’m closing this ticket. If you run into missing pieces while using pixi, please reopen with details.

closed 2025-11-03 by jslee02