dartsim/dart #1474: HPP-FCL

costashatz, 2020-06-16:
I recently came along [hpp-fcl](https://github.com/humanoid-path-planner/hpp-fcl) which seems to be an extension of FCL that fixes various bugs, adds python bindings and a few more.

I guess it is not hard to switch to this library instead of FCL, right? Does this library have the features that are missing from FCL (and makes DART convert objects to meshes?)? It might be a good idea to switch to this one..

jslee02, 2020-09-18:
My biggest concern is that FCL is also improving mainly by TRI so two versions could diverse. It'd be ideal if we can port the fixes from hpp-fcl to FCL instead.

costashatz, 2020-09-18:
This makes sense. I had a deeper look at fcl and hpp-fcl, and I think fcl is becoming superior. I do not know all the small details, but we could definitely have a look at the bug fixes hpp-fcl did and if they still are relevant for fcl.

Btw, is the issue of fcl with basic shapes fixed in newer versions of fcl/DART?

jslee02, 2025-11-03:
> Btw, is the issue of fcl with basic shapes fixed in newer versions of fcl/DART?

I don't think so unfortunatley, which should be fixed by upstream. Closing as we have no plan to migrate to hpp-fcl.

closed 2025-11-03 by jslee02