dartsim/dart #1475: Question about the difference between "skel->getVelocities() to get root linear velocity" and "skel->getBodyNode(0)->getCOMLinearVelocity()""

zhouyan09, 2020-06-25:
Hi, I build a skeleton(marked as "skel" below) with "Torso of FreeJoint" and 17 other Joints of "BallJoint", but I am confused by the difference in the definition of "root velocity", can someone help me figure that out:I can get the root velocity using "skel->getRootBodyNode()->getCOMLinearVelocity() ", but the value is different from the function"skel->getCOMLinearVelocity()" or the value extracted from the function "skel->getVelocities()" .
So, can someone help me with that, which one is the correct "root velocity"? What are the differences among the three function :"skel->getCOMLinearVelocity()", "skel->getVelocities()" and "skel->getRootBodyNode()->getCOMLinearVelocity() ".
Thank you very much!

mxgrey, 2020-06-25:
The root velocity refers to the velocity of the `BodyNode` (a.k.a. rigid body or link) that represents the "root" of the skeleton's tree. A root is simply a `BodyNode` that does not have a parent `BodyNode`.

When you do `skel->getRootBodyNode()->getCOMLinearVelocity()` you are getting the linear velocity of the center of mass of *just the root body node*. This will ignore the velocities of any other bodies in the skeleton.

When you do `skel->getCOMLinearVelocity()` you are getting the linear velocity of the center of mass of the overall skeleton, taking all `BodyNode`s into account.

When you do `skel->getVelocities()` you are getting the **generalized velocities** of the skeleton. That's a vector of length `N` where `N` is the number of degrees of freedom in the skeleton. You can think of this as a vector of joint velocities.

zhouyan09, 2020-06-28:
> The root velocity refers to the velocity of the `BodyNode` (a.k.a. rigid body or link) that represents the "root" of the skeleton's tree. A root is simply a `BodyNode` that does not have a parent [...]

Thank you for your answer. There are still some points I want to make clear: the function  "skel->getVelocities()" gives out the joint velocities, and they are different from "velocities of the center of mass of the nodes", is it right?
I set the position vector of the skeleton using the "root global positions(3D vector) and all the body node joint Axis-angle(3D)", when I get the vector `skel->getVelocities()`, the first three elements do not equal to `skel->getRootBodyNode()->getCOMLinearVelocity()`, for the reason of the difference between root joint velocities and the velocities of the center of mass of the root node, am I right? If I misunderstood, the value of "skel->getRootBodyNode()->getCOMLinearVelocity()" should be part of the vector from the function `skel->getVelocities()`. Can you explain more details about that? Thank you very much.

mxgrey, 2020-06-28:
Generalized velocities (which are returned by `getVelocities()` are the time derivatives of [generalized coordinates](https://en.m.wikipedia.org/wiki/Generalized_coordinates). Generalized velocities will not necessarily be the same as linear and angular velocities that are used when describing the velocity of a single rigid body.

In particular, I believe DART uses [Lie algebra](https://en.m.wikipedia.org/wiki/Exponential_map_(Lie_theory)) representations for the generalized velocities of `FreeJoint` and `BallJoint`.