dartsim/dart #1476: Linking issues on Visual Studio 2019

Robert-Lu, 2020-07-09:
> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [ **X** ] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [ **X** ] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: [e.g., master, 6.8.3]
-- **6.9.2**
* OS name and version name(or number): [e.g., Ubuntu 18.04, macOS Mojav, Windows 10]
-- **Windows 10**
* Compiler name and version number: [e.g., GCC 7.4.0, Clang 3.9.1]
-- **Visual Studio 2019 (MSVC 142)**

#### Expected Behavior
> Please describe the behavior you are expecting.

Compile the hello-world example, after successfully installed DART on Windows following the [guide](https://dartsim.github.io/install_dart_on_windows.html).

#### Current Behavior
> What is the current behavior?
> If you have trouble in `cmake` or `make`, it would be very helpful to build DART with verbose option:
> `$ cmake .. -DDART_VERBOSE=On`
> `$ make VERBOSE=1`
> and then dump the results into a [gist](https://gist.github.com/) and share the link to gist here.

##### Before this Issue: 

I met some other issues to compile the VS solution that CMake generates for the DART examples. 

* The installation of DART is swift and fluent. 
  * Boost version is 1.72.0, caused some warnings.
  * OSG version is 3.6.4, and is built from source previously. 
* Template issue that cannot compile, as described in this [issue](https://github.com/dartsim/dart/issues/753) (#753). It is resolved after adding the `/permissive-` flag.
* Some other MACRO issues. Resolved by adding the definition of `_USE_MATH_DEFINES `. (as this [post](https://stackoverflow.com/questions/26065359/m-pi-flagged-as-undeclared-identifier) tells)
* Linking errors because some libraries were not added by CMake. In my case `assimp`, `fcl` and `osgUtil` are not in the `Input` field of linker property.

##### Linker Issue:

After resolved all the issues above, there are still 7 unresolved externals, which caused LNK1120 error.

The unresolved external symbols are:
```
__declspec(dllimport) public: __cdecl dart::gui::osg::TrackballManipulator::TrackballManipulator(int)" (__imp_??0TrackballManipulator@osg@gui@dart@@QEAA@H@Z)
__declspec(dllimport) public: void __cdecl dart::gui::osg::TrackballManipulator::`vbase destructor'(void)" (__imp_??_DTrackballManipulator@osg@gui@dart@@QEAAXXZ)
__declspec(dllimport) public: virtual char const * __cdecl dart::gui::osg::TrackballManipulator::className(void)const " (__imp_?className@TrackballManipulator@osg@gui@dart@@UEBAPEBDXZ)
__declspec(dllimport) public: virtual char const * __cdecl dart::gui::osg::TrackballManipulator::libraryName(void)const " (__imp_?libraryName@TrackballManipulator@osg@gui@dart@@UEBAPEBDXZ)
__declspec(dllimport) public: virtual bool __cdecl dart::gui::osg::TrackballManipulator::isSameKindAs(class osg::Object const *)const " (__imp_?isSameKindAs@TrackballManipulator@osg@gui@dart@@UEBA_NPEBVObject@2@@Z)
__declspec(dllimport) public: virtual class osg::Object * __cdecl dart::gui::osg::TrackballManipulator::clone(class osg::CopyOp const &)const " (__imp_?clone@TrackballManipulator@osg@gui@dart@@UEBAPEAVObject@2@AEBVCopyOp@2@@Z)
__declspec(dllimport) public: virtual class osg::Object * __cdecl dart::gui::osg::TrackballManipulator::cloneType(void)const " (__imp_?cloneType@TrackballManipulator@osg@gui@dart@@UEBAPEAVObject@2@XZ)
```

A more complete error information for the first of it is as follows: (`dart-gui-osg.lib(Viewer.obj)` might be informative)
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2001	unresolved external symbol "__declspec(dllimport) public: __cdecl dart::gui::osg::TrackballManipulator::TrackballManipulator(int)" (__imp_??0TrackballManipulator@osg@gui@dart@@QEAA@H@Z)	hello_world	C:\dev\dart-6.9.2\examples\build\hello_world\dart-gui-osg.lib(Viewer.obj)	1	
```

All are related to `dart::gui::osg::TrackballManipulator` and this is strange. I have used `dumpbin` to check the symbols, and in all of the dart library files, there are related symbols in `dart-gui-osg.lib` (the RelWithDebInfo version, but I compile with Release version), and it contains related symbols, as follows:

``` bash
>>> dumpbin /headers dart-gui-osg.lib | grep dart::gui::osg::TrackballManipulator
         COMDAT; sym= "[thunk]:public: virtual void * __cdecl dart::gui::osg::TrackballManipulator::`vector deleting destructor'`vtordisp{4294967292,0}' (unsigned int)" (??_ETrackballManipulator@osg@gui@dart@@$4PPPPPPPM@A@EAAPEAXI@Z)
         COMDAT; sym= "public: virtual void * __cdecl dart::gui::osg::TrackballManipulator::`scalar deleting destructor'(unsigned int)" (??_GTrackballManipulator@osg@gui@dart@@UEAAPEAXI@Z)
         COMDAT; sym= "[thunk]:public: virtual char const * __cdecl dart::gui::osg::TrackballManipulator::className`vtordisp{4294967292,0}' (void)const " (?className@TrackballManipulator@osg@gui@dart@@$4PPPPPPPM@A@EBAPEBDXZ)
         COMDAT; sym= "[thunk]:public: virtual class osg::Object * __cdecl dart::gui::osg::TrackballManipulator::clone`vtordisp{4294967292,0}' (class osg::CopyOp const &)const " (?clone@TrackballManipulator@osg@gui@dart@@$4PPPPPPPM@A@EBAPEAVObject@2@AEBVCopyOp@2@@Z)
         COMDAT; sym= "[thunk]:public: virtual class osg::Object * __cdecl dart::gui::osg::TrackballManipulator::cloneType`vtordisp{4294967292,0}' (void)const " (?cloneType@TrackballManipulator@osg@gui@dart@@$4PPPPPPPM@A@EBAPEAVObject@2@XZ)
         COMDAT; sym= "[thunk]:public: virtual bool __cdecl dart::gui::osg::TrackballManipulator::isSameKindAs`vtordisp{4294967292,0}' (class osg::Object const *)const " (?isSameKindAs@TrackballManipulator@osg@gui@dart@@$4PPPPPPPM@A@EBA_NPEBVObject@2@@Z)
         COMDAT; sym= "[thunk]:public: virtual char const * __cdecl dart::gui::osg::TrackballManipulator::libraryName`vtordisp{4294967292,0}' (void)const " (?libraryName@TrackballManipulator@osg@gui@dart@@$4PPPPPPPM@A@EBAPEBDXZ)
         COMDAT; sym= "const dart::gui::osg::TrackballManipulator::`local vftable'{for `osg::Object'}" (??_STrackballManipulator@osg@gui@dart@@6BObject@1@@)
         COMDAT; sym= "class dart::gui::osg::TrackballManipulator `RTTI Type Descriptor'" (??_R0?AVTrackballManipulator@osg@gui@dart@@@8)
         COMDAT; sym= "const dart::gui::osg::TrackballManipulator::`RTTI Complete Object Locator'{for `osg::Object'}" (??_R4TrackballManipulator@osg@gui@dart@@6BObject@1@@)
         COMDAT; sym= "dart::gui::osg::TrackballManipulator::`RTTI Class Hierarchy Descriptor'" (??_R3TrackballManipulator@osg@gui@dart@@8)
         COMDAT; sym= "dart::gui::osg::TrackballManipulator::`RTTI Base Class Array'" (??_R2TrackballManipulator@osg@gui@dart@@8)
         COMDAT; sym= "dart::gui::osg::TrackballManipulator::`RTTI Base Class Descriptor at (0,-1,0,64)'" (??_R1A@?0A@EA@TrackballManipulator@osg@gui@dart@@8)
         COMDAT; sym= "public: __cdecl dart::gui::osg::TrackballManipulator::TrackballManipulator(class dart::gui::osg::TrackballManipulator const &,class osg::CopyOp const &)" (??0TrackballManipulator@osg@gui@dart@@QEAA@AEBV0123@AEBVCopyOp@1@@Z)
         COMDAT; sym= "int `public: __cdecl dart::gui::osg::TrackballManipulator::TrackballManipulator(class gui::osg::TrackballManipulator::dtor$0 const &,class TrackballManipulator::CopyOp const &)'::`1'::dtor$0" (?dtor$0@?0???0TrackballManipulator@osg@gui@dart@@QEAA@AEBV0123@AEBVCopyOp@1@@Z@4HA)
         COMDAT; sym= "int `public: __cdecl dart::gui::osg::TrackballManipulator::TrackballManipulator(class gui::osg::TrackballManipulator::dtor$1 const &,class TrackballManipulator::CopyOp const &)'::`1'::dtor$1" (?dtor$1@?0???0TrackballManipulator@osg@gui@dart@@QEAA@AEBV0123@AEBVCopyOp@1@@Z@4HA)
         COMDAT; sym= "public: __cdecl dart::gui::osg::TrackballManipulator::TrackballManipulator(int)" (??0TrackballManipulator@osg@gui@dart@@QEAA@H@Z)
         COMDAT; sym= "int `public: __cdecl dart::gui::osg::TrackballManipulator::TrackballManipulator(int)'::`1'::dtor$0" (?dtor$0@?0???0TrackballManipulator@osg@gui@dart@@QEAA@H@Z@4HA)
         COMDAT; sym= "int `public: __cdecl dart::gui::osg::TrackballManipulator::TrackballManipulator(int)'::`1'::dtor$1" (?dtor$1@?0???0TrackballManipulator@osg@gui@dart@@QEAA@H@Z@4HA)
[... cut ...] art@@UEAA_NNNN@Z)
         COMDAT; sym= "public: virtual bool __cdecl dart::gui::osg::TrackballManipulator::performMovementRightMouseButton(double,double,double)" (?performMovementRightMouseButton@TrackballManipulator@osg@gui@dart@@UEAA_NNNN@Z)
```

P.S. I also tried with the config RelWithDebInfo, the same.

Thank you for your help. And maybe I would better try to develop under Linux. I can get everything works on Linux but I have to remotely use it because of the pandemic. Everyone, keep safe.

Tomohiro-Nagasaka, 2021-02-04:
When building libraries and hello-world, I added OSG_LIBRARY_STATIC to preprocessor macro. I could build but there still is an error on runtime reported here: https://github.com/dartsim/dart/issues/1431

jslee02, 2021-04-14:
I believe this issue is fixed by recent build fixes for Windows. Please reopen if that's not the case.

closed 2021-04-14 by jslee02