dartsim/dart #1482: Fails to build with GCC-10 on riscv64

cpaelzer, 2020-08-11:
#### Environment

* DART version: [e.g., master, 6.9.2]
* OS Ubuntu 20.04 + 20.10 (Also Debian)
* Compiler name and version number: GCC-10 (worked with GCC-9)

#### Expected Behavior
Build and tests work fine

#### Current Behavior
On s390x the build time tests fail
```
	 29 - test_ContactConstraint (SEGFAULT)
```

Here the initial [build log](https://launchpadlibrarian.net/492858051/buildlog_ubuntu-groovy-riscv64.dart_6.9.2-3build2_BUILDING.txt.gz)

I have rebuilt the same with verbosity as recommended - [the build is in this PPA](https://launchpad.net/~ci-train-ppa-service/+archive/ubuntu/4191/+packages) and you can access the [logs for all riscv64](https://launchpad.net/~ci-train-ppa-service/+archive/ubuntu/4191/+build/19812925) from there.

#### Steps to Reproduce

1. Get dart sources and install build dependencies
2. get GCC-10
3. build on RiscV64
3. run tests

jslee02, 2020-08-12:
Thank you for the report! I'm not familiar with the architectures (riscv64 and s390x). Could you please let me know where these architectures are generally used for? Also, I'm afraid that I'm not sure how to reproduce and debug the issues without having those architectures. Is there a way to emulate them?

cpaelzer, 2020-08-12:
Hi,
they are on opposite end of the architecture spectrum.

 * s390x is known as mainframe, it is an architecture mostly used for banking and databases, but great in general - see https://en.wikipedia.org/wiki/IBM_Z
 * riscv64 on the other hand you can think of as arm, but without license fees https://en.wikipedia.org/wiki/RISC-V

s390x is hard to emulate right, but you can get access to a free virtual guest at https://developer.ibm.com/components/ibm-linuxone/gettingstarted/

riscv64 can be emulated via qemu like:
```
qemu-system-riscv64 -machine virt -m 2048 -smp 4 \
  -bios qemu-virt-20200504-ubuntu-firmware.bin \
  -device virtio-blk-device,drive=vda \
  -drive file=grovy-updated-20200811.qcow2,id=vda \
  -device virtio-net-device,netdev=eth0 \
  -netdev user,id=eth0,hostfwd=tcp::5555-:22
```

You can get instructions and the base images (for Ubuntu 20.04) here https://people.ubuntu.com/~wgrant/riscv64/

cpaelzer, 2020-08-14:
I was slowly building this on a local riscv64 qemu emulation, I got to the same crash.
For the sake of being generally usable I was testing another test:

```
root@ubuntu:~/dart-6.9.2# ./obj-riscv64-linux-gnu/unittests/unit/test_CollisionGroups
Running main() from /root/dart-6.9.2/unittests/gtest/src/gtest_main.cc
[==========] Running 8 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 8 tests from CollisionEngine/CollisionGroupsTest
[ RUN      ] CollisionEngine/CollisionGroupsTest.SkeletonSubscription/0
Running CollisionGroups test for [dart]
[       OK ] CollisionEngine/CollisionGroupsTest.SkeletonSubscription/0 (78 ms)
...
[       OK ] CollisionEngine/CollisionGroupsTest.BodyNodeSubscription/3 (0 ms)
[----------] 8 tests from CollisionEngine/CollisionGroupsTest (194 ms total)

[----------] Global test environment tear-down
[==========] 8 tests from 1 test case ran. (198 ms total)
[  PASSED  ] 8 tests.
```

So assuming that I can generally use this for debugging I was running the failing test and it crashed as expected.
```
root@ubuntu:~/dart-6.9.2# ./obj-riscv64-linux-gnu/unittests/unit/test_ContactConstraint 
Running main() from /root/dart-6.9.2/unittests/gtest/src/gtest_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from ContactConstraint
[ RUN      ] ContactConstraint.ContactWithKinematicJoint
Segmentation fault
```

Throwing the same into GDB is unfortunately far from helpful:
```
Starting program: /root/dart-6.9.2/obj-riscv64-linux-gnu/unittests/unit/test_ContactConstraint 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/riscv64-linux-gnu/libthread_db.so.1".
Running main() from /root/dart-6.9.2/unittests/gtest/src/gtest_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from ContactConstraint
[ RUN      ] ContactConstraint.ContactWithKinematicJoint

Program received signal SIGSEGV, Segmentation fault.
0x0000000000001000 in ?? ()
(gdb) bt
#0  0x0000000000001000 in ?? ()
(gdb) 
```

I was starting at `testContactWithKinematicJoint` and stepping my way towards the problem.
It is a bit hard to track that way around, but once crashed the backtrace is empty.

The last hit of any of "getLinearVelocity,getSpatialVelocity,getTransform" is `const Eigen::Vector6d& Frame::getSpatialVelocity() const`.

Then next I went deeper from there of the list "getSpatialVelocity,getRelativeTransform,AdInvT,getRelativeSpatialVelocity,getParentFrame" the last hit is   `dart::math::AdInvT (_T=..., _V=...) at ./dart/math/Geometry.cpp:714`

I checked dart::math::AdInvT in a multiple runs and it was hit 5582 times before the crash every time. That allows to step onto the failing execution of AdInvT.

Stack at this point looks like
```
(gdb) bt
#0  0x0000003ff7d6ffe8 in dart::dynamics::Frame::getSpatialVelocity (this=0x2aaab2e768) at ./dart/dynamics/Frame.cpp:619
#1  dart::dynamics::Frame::getSpatialVelocity (this=0x2aaab2e768) at ./dart/dynamics/Frame.cpp:136
#2  0x0000003ff7d71996 in dart::dynamics::Frame::getSpatialVelocity (this=this@entry=0x2aaab2e768, _relativeTo=_relativeTo@entry=0x3ff7fe0318 <dart::dynamics::Frame::World()::world>, 
    _inCoordinatesOf=0x3ff7fe0318 <dart::dynamics::Frame::World()::world>) at ./dart/dynamics/Frame.cpp:168
#3  0x0000003ff7d71aca in dart::dynamics::Frame::getLinearVelocity (this=this@entry=0x2aaab2e768, _relativeTo=_relativeTo@entry=0x3ff7fe0318 <dart::dynamics::Frame::World()::world>, 
    _inCoordinatesOf=<optimized out>) at ./dart/dynamics/Frame.cpp:222
#4  0x0000002aaaacb390 in testContactWithKinematicJoint (lcpSolver=..., tol=9.9999999999999995e-07) at ./unittests/unit/test_ContactConstraint.cpp:88
#5  0x0000002aaaacba42 in ContactConstraint_ContactWithKinematicJoint_Test::TestBody (this=<optimized out>) at /usr/include/c++/10/new:175
[... 157 lines cut ...]
a3             0x1	1
a4             0x5a6359d952f93000	6513148276042706944
a5             0x5a6359d952f93000	6513148276042706944
a6             0x0	0
a7             0x2aaab2d1d8	183252472280
s2             0x3fffffef38	274877902648
s3             0x3ff7ffd8a8	274743679144
s4             0x2aaaaf29c8	183252232648
s5             0x2aaaad0e0c	183252094476
s6             0x0	0
s7             0x2aaab207e0	183252420576
s8             0x1	1
s9             0x0	0
s10            0x2aaab0a234	183252329012
s11            0x1	1
t3             0x3ff6ea9dc6	274725510598
t4             0x15	21
t5             0x1	1
t6             0x10	16
pc             0x2aaaacb554	0x2aaaacb554 <testContactWithKinematicJoint(std::shared_ptr<dart::constraint::BoxedLcpSolver> const&, double)+5146>
```

It seems like the program flow itself was broken, register `t1` holds that suspicious 0x1000 but I'm not an riscv64 expert.

Might after all be a gcc-10 bug as it seemed to work fine with gcc-9 (or it now exposes a bug in dart). I can't see an immediate fix out of that, but I hope it helps you to see what might be going on.

jslee02, 2025-11-03:
Thanks for the detailed report! Unfortunately we don’t officially support that platform, so we don’t have a fix planned. We’d gladly review a PR if you’re able to add or repair support on that environment.

closed 2025-11-03 by jslee02