dartsim/dart #1483: Fails to build on s390x

cpaelzer, 2020-08-11:
#### Environment

* DART version: [e.g., master, 6.9.2]
* OS Ubuntu 20.04 + 20.10 (Also Debian)
* Compiler name and version number: GCC-9 + GCC-10

#### Expected Behavior
Build and tests work fine

#### Current Behavior
On s390x the build time tests fail
```
	 17 - test_ForwardKinematics (SEGFAULT)
	 47 - test_SdfParser (SEGFAULT)
	 50 - test_DartLoader (SEGFAULT)
	 51 - test_IkFast (SEGFAULT)
```
Here the initial [build log](https://launchpadlibrarian.net/490693560/buildlog_ubuntu-focal-s390x.dart_6.9.2-2build4_BUILDING.txt.gz)

I have rebuilt the same with verbosity as recommended - [the build is in this PPA](https://launchpad.net/~ci-train-ppa-service/+archive/ubuntu/4191/+packages) and you can access the [logs for s390x](https://launchpad.net/~ci-train-ppa-service/+archive/ubuntu/4191/+build/19812926) from there.

#### Steps to Reproduce

1. Get dart sources and install build dependencies
2. build on s390x
3. run tests

cpaelzer, 2020-08-12:
Note - s390x seems to have never built AFAICS

jslee02, 2025-11-03:
Thanks for the detailed report! Unfortunately we don’t officially support that platform, so we don’t have a fix planned. We’d gladly review a PR if you’re able to add or repair support on that environment.

closed 2025-11-03 by jslee02