dartsim/dart #1484: Mixed intrinsic and extrinsic camera parameter in OpenGL projection matrix

zhengpingzhou, 2020-08-14:
### Bug Report

- [ x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [ x] I checked to make sure that this issue has not already been filed.

#### Environment

* DART version: 6.3.0
* OS name and version name(or number): Ubuntu 18.04
* Compiler name and version number: GCC 7.5.0

#### Expected Behavior

The OpenGL projection matrix is a 4x4 matrix that describes the camera intrinsic parameters. It is expected to have the following format:

[image]

Therefore, if you print the value of `GL_PROJECTION_MATRIX`, the 16th element must be zero.

#### Current Behavior

> What is the current behavior?

Currently, if you construct a `SimWindow` following the default setup, the 16th element of `GL_PROJECTION_MATRIX` is not zero. For example, when the aspect ratio is 4:3, the projection matrix is:

```
 1.81066        0        0        0
       0  2.41421        0        0
       0        0  -1.0202 0.818182
       0        0       -1        1
```

It seems the cause of the problem is that part of extrinsic parameters (which are supposed to be included in the model-view matrix) is integrated into the projection matrix (which is supposed to only contain the intrinsic parameters):

https://github.com/dartsim/dart/blob/v6.3.0/dart/gui/Win3D.cpp#L169

This may lead to confusion if the camera parameters need to be exported and used in other programs. Although this does not expose any rendering problem, I personally suggest moving the `gluLookAt` line into the initialization section for the model-view matrix. Kindly let me know if I'm missing any context.

#### Code to Reproduce

[dart-opengl-projection-matrix-bug-report](https://gist.github.com/zhengpingzhou/0dc1a8a0f39a86e564d86740f6ea2f42)

#### Steps to Reproduce

Build and run the code. The code prints the value of `GL_PROJECTION_MATRIX` when the aspect ratio is 4:3.

####

jslee02, 2020-08-16:
Thanks for the report with the test code! I was able to reproduce the incorrect projection matrix computation and fix it by moving the `gluLookAt` to the model-view section, as you suggested (#1485).

zhengpingzhou, 2020-08-16:
Great, thanks!

PR #1485 (2020-08-16): Fix mixed intrinsic and extrinsic camera parameters in OpenGL projection matrix
This PR fixes the incorrect projection matrix computation in OpenGL due to the misplaced `gluLookAt()` calls.

Fixes #1484.

***

**Before creating a pull request**

- [x] (N/A) Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [x] (N/A) Add unit test(s) for this change
closed 2020-08-16 by jslee02