dartsim/dart #1492: error: no template named 'void_t' in namespace 'std'; did you mean 'boost::void_t'? (etc)

yurivict, 2020-08-28:
Version ```6.9.3``` breaks:

```
In file included from /usr/ports/misc/dartsim/work/dart-6.9.3/dart/optimizer/pagmo/PagmoMultiObjectiveProblemAdaptor.cpp:33:
In file included from /usr/ports/misc/dartsim/work/dart-6.9.3/dart/optimizer/pagmo/PagmoMultiObjectiveProblemAdaptor.hpp:37:
In file included from /usr/local/include/pagmo/pagmo.hpp:35:
In file included from /usr/local/include/pagmo/algorithm.hpp:46:
In file included from /usr/local/include/pagmo/detail/typeid_name_extract.hpp:36:
/usr/local/include/pagmo/type_traits.hpp:55:26: error: no template named 'void_t' in namespace 'std'; did you mean 'boost::void_t'?
struct detector<Default, std::void_t<Op<Args...>>, Op, Args...> {
                         ^~~~~
/usr/local/include/boost/type_traits/make_void.hpp:25:1: note: 'boost::void_t' declared here
using void_t = typename make_void<Ts...>::type;
^
In file included from /usr/ports/misc/dartsim/work/dart-6.9.3/dart/optimizer/pagmo/PagmoMultiObjectiveProblemAdaptor.cpp:33:
In file included from /usr/ports/misc/dartsim/work/dart-6.9.3/dart/optimizer/pagmo/PagmoMultiObjectiveProblemAdaptor.hpp:37:
In file included from /usr/local/include/pagmo/pagmo.hpp:35:
In file included from /usr/local/include/pagmo/algorithm.hpp:46:
In file included from /usr/local/include/pagmo/detail/typeid_name_extract.hpp:36:
/usr/local/include/pagmo/type_traits.hpp:72:26: error: no template named 'conjunction' in namespace 'std'
using conjunction = std::conjunction<Args...>;
                    ~~~~~^
/usr/local/include/pagmo/type_traits.hpp:75:26: error: no template named 'disjunction' in namespace 'std'; did you mean 'is_function'?
using disjunction = std::disjunction<Args...>;
                    ~~~~~^
```

yurivict, 2020-08-28:
You are missing ```--std=c++17``` because std::void_t (through pagmo2) is a c++17 feature.

jslee02, 2020-08-28:
Thank you for the report.

It seems [pagmo2 recently added the C++17 requirement in their master branch]((https://github.com/esa/pagmo2/commit/228332727377a627fe95a4f471f30a453e279fbe#diff-af3b638bc2a3e6c650974192a53c7291)), and building DART with the latest release of pagmo2 (i.e. v2.15.0) should work fine. Specifically, `std::void_t`, `std::conjuction`, `std::disjunction` are added in this [commit](https://github.com/esa/pagmo2/commit/572286270a91c3cc4f3709700c59d547d3b13d73).

Maybe DART needs to check the detected pagmo2 version and add the required compiler flags for C++17 for pagmo2 > v2.15.0.

closed 2021-07-25 by jslee02