dartsim/dart #1499: Visualisation with plotly / plotly express

PitS99, 2020-09-16:
Is there any way, how to do a **visualisation** of a forecast (and original data) **with plotly or plotly express** instead of matplotlib?

from darts.models import ExponentialSmoothing
train, val = series.split_before(pd.Timestamp('2020-05-01'))
model = ExponentialSmoothing()
model.fit(train)
prediction = model.predict(len(val))

PitS99, 2020-09-16:
Is there any way to store the forecast values to dataframe?

jslee02, 2020-09-16:
I believe you refer to [Darts](https://unit8co.github.io/darts/index.html) that is not this repository is for. Maybe you want to ask your question to [their repo](https://github.com/unit8co/darts/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc).

closed 2020-09-16 by jslee02