dartsim/dart #1501: Support BUILD_SHARED_LIBS set to ON on Windows/MSVC

traversaro, 2020-09-17:
**Is your feature request related to a problem? Please describe.**
Given its huge size, for some applications it is convenient to be able to compiled dart as a shared library on Windows/MSVC, to permit to share it among multiple executable.

**Additional context**
Discussion on this was ongoing on https://github.com/dartsim/dart/issues/1005, but that issue was closed as its main objective was reached without the need to make sure that DART supported being compiled as a shared library on Windows.

jslee02, 2020-09-18:
Do you know what's preventing us from building DART as shared libraries? Doesn't `BUILD_SHARED_LIBS` work as it is?

traversaro, 2020-09-18:
To be honest I did not tried myself, but from https://github.com/dartsim/dart/issues/1005 it seems that the situation is the following: 
* The CMake variable `CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS` does not work as the library has more then 65k symbols, see https://github.com/dartsim/dart/issues/1005#issuecomment-375144278 . 
* An attempt was made in https://github.com/dartsim/dart/pull/1040 to add explicit visibility macros, but i guess there were some problem/doubts on where to add visibility macros for heavily templated class, and then eventually the PR was not completed.

jslee02, 2023-01-14:
Fixed by d78c3286ee371fe544f93bc2c2040d550da18685

closed 2023-01-14 by jslee02

traversaro, 2023-01-14:
Great, thanks a lot!

jslee02, 2023-01-14:
Thanks for being patient with this. I had to implement some workarounds and make adjustments to the code to get it working. I will be addressing these issues and making improvements in future commits.