dartsim/dart #1507: Dartpy importing Issue on virtualenv

shbang91, 2020-09-23:
#### Environment
> Select the following information.
* DART version: dartpy
* OS name and version name(or number): macOS Catalina
* Compiler : Clang 11.0
* I created a virtual environment using pip3 install virtualenv and it uses my system python 3.7.

#### Current Behavior
> What is the current behavior?
After pip3 install dartpy, I can find a dartpy.so in the lib directory in my virtual environment. However, when I try importing dartpy in python, e.g., import dartpy, it spits out the segmentation fault error.

#### Steps to Reproduce
> Please provide detailed steps for reproducing the issue.
1. Create a virtual environment using pip3 install virtualenv
2. Install dartpy by using pip3 install dartpy
3. Using python, import dartpy
4. You'll get the segmentation fault on your screen

jslee02, 2020-09-23:
Quick check: Have you installed all the dependencies?
http://dartsim.github.io/install_dartpy_on_macos.html#install-dartpy-using-pip3

shbang91, 2020-09-23:
Yeah, I double-checked all the dependencies, but it still shows segmentation fault when importing dartpy library.

jslee02, 2020-09-23:
Hm, I wasn't able to reproduce the segfault on my macOS (Catalina). Here is what I see when installing `dartpy`:

```
(venv) ➜  tmp10 pip3 install dartpy -U
Collecting dartpy
  Downloading dartpy-0.0.1.post11.tar.gz (1.4 MB)
     |████████████████████████████████| 1.4 MB 1.0 MB/s
Building wheels for collected packages: dartpy
  Building wheel for dartpy (setup.py) ... done
  Created wheel for dartpy: filename=dartpy-0.0.1.post11-cp38-cp38-macosx_10_14_x86_64.whl size=2080 sha256=87033e8319d71ebec1d1495ddf39a13a5a77fba01729a983096656744d8f69a8
  Stored in directory: /Users/jeoslee/Library/Caches/pip/wheels/cb/33/d9/ccc9b71a55c7939af21d354dfc64fa09ab2f814e8c728cdc0a
Successfully built dartpy
Installing collected packages: dartpy
Successfully installed dartpy-0.0.1.post11
WARNING: You are using pip version 20.2.2; however, version 20.2.3 is available.
You should consider upgrading via the '/Users/jeoslee/tmp10/venv/bin/python -m pip install --upgrade pip' command.
(venv) ➜  tmp10 python
Python 3.8.5 (default, Jul 21 2020, 10:42:08)
[Clang 11.0.0 (clang-1100.0.33.17)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import dartpy
>>> dartpy.__file__
'/Users/jeoslee/tmp10/venv/lib/python3.8/site-packages/dartpy.so'
```

Do you see any error message when it segfaults?

shbang91, 2020-09-23:
It just shows segmentation fault: 11 as below:

```
[pypnc] seunghyeonbang ~ python
Python 3.7.3 (default, Dec 13 2019, 19:58:14)
[Clang 11.0.0 (clang-1100.0.33.17)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import dartpy
Segmentation fault: 11
```
So, I was trying to uninstall dartpy by doing pip3 uninstall dartpy, but it didn't completely remove dartpy.so, which it leads to incorrect reinstalling as you mentioned on: https://dartsim.github.io/install_dartpy_on_macos.html.
Do you have any suggestion?

jslee02, 2020-09-24:
What do you see if you run Python with `-v` option and try to import `dartpy`?:
```
[pypnc]seunghyeonbang ~ python -v
>>> import dartpy
```

To uninstall, you could safely remove `dartpy.so` manually (by running `sudo rm <path_to_dartpy.so>/dartpyso` as it's the only file installed (without changing any other system configurations).

shbang91, 2020-09-24:
I manually removed dartpy.so in the lib directory and tried to reinstall dartpy by doing pip3 install dartpy. But, I don't see dartpy.so in the lib directory so I cannot import dartpy right now as you suggested. Here is the result:
`[pypnc] seunghyeonbang ~ pip3 install dartpy
Requirement already satisfied: dartpy in ./.venv/pypnc/lib/python3.7/site-packages (0.0.1.post11)`

closed 2025-11-13 by jslee02