dartsim/dart #1524: Error in CapsuleShape inertia calculation

azeey, 2020-11-06:
> If you're not reporting a bug, please use the [forum](https://dartsim.discourse.group/) to ask questions.

### Bug Report
> Please answer the following questions for yourself before reporting a bug.
- [x] I checked the [documentation](http://dartsim.github.io/) and the [forum](https://dartsim.discourse.group/) but found no answer.
- [x] I checked to make sure that this issue has not already been filed.

#### Environment
> Select the following information.
* DART version: master
* OS name and version name(or number): all
* Compiler name and version number: all

#### Current Behavior
The inertia calculation for `CapsuleShape` appears to be missing a `1/4` coefficient for the `Ixx` term.
https://github.com/dartsim/dart/blob/4b0b4dcedae6c6f0de50a1d3d13015b178aa68bb/dart/dynamics/CapsuleShape.cpp#L124-L126

The referenced website in the code shows:
[image]

, but following the formulation shows that the `H^2/2` in term is actually supposed to be `(H/2)^2`. This is also stated in one of the comments on the website. And the `2m_hs` term is equal to `massSphere`. So I believe the code should be

```c++
+ massSphere 
       * ((1.0 /4.0) * height2 + (3.0 / 8.0) * height * radius 
          + (2.0 / 5.0) * radius2); 
```

scpeters, 2020-11-07:
See also the discussion at https://github.com/ignitionrobotics/ign-math/pull/163/files#r517544323

jslee02, 2021-04-15:
Thanks for the fix! This should be resolved by https://github.com/dartsim/dart/pull/1561.

PR #1561 (2021-04-15): Fix inertia calculation of CapsuleShape
Fixes #1524 

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [ ] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change
closed 2021-04-15 by jslee02