dartsim/dart #1525: setting up dartpy with conda environment

dgtim, 2020-11-09:
Hi, while trying to set up / build the dartpy on the computational cluster in my own work folder with no root privileges or alike. Here is the issue:

Building my own conda environment first:

```
module load miniconda
conda create -n mydartpyenv
conda activate mydartpyenv
module load cmake
conda install -c omnia eigen3
conda install -c conda-forge libccd fcl assimp urdfdom openscenegraph pybind11
```

Then building the dartpy:

```
git clone git://github.com/dartsim/dart.git
cd dart
git checkout tags/v6.9.0
mkdir build
cd build
cmake .. -DDART_BUILD_DARTPY=ON -DCMAKE_INSTALL_PREFIX=$WRKDIR/dartpy -DCMAKE_BUILD_TYPE=Release
make -j4 dartpy
make install
```

that should work in theory, but in practice the last cmake gets stuck with the missing fcl package while it is there.

```
(mydartpyenv) [build]$ cmake .. -DDART_BUILD_DARTPY=ON -DCMAKE_INSTALL_PREFIX=$WRKDIR/dartpy -DCMAKE_BUILD_TYPE=Release

...
CMake Error at /share/apps/spack/envs/fgci-centos7-generic/software/cmake/3.15.3/qvh6hn6/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find fcl (missing: FCL_INCLUDE_DIRS FCL_LIBRARIES) (Required is
  at least version "0.3.2")
Call Stack (most recent call first):
  /share/apps/spack/envs/fgci-centos7-generic/software/cmake/3.15.3/qvh6hn6/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/Findfcl.cmake:60 (find_package_handle_standard_args)
  cmake/DARTFindfcl.cmake:9 (find_package)
  cmake/DARTMacros.cmake:81 (include)
  cmake/DARTFindDependencies.cmake:21 (dart_find_package)
  CMakeLists.txt:148 (include)
```

setting both vars manually did not help:

```
export FCL_LIBRARIES=$HOME/.conda/envs/mydartpyenv/lib
export FCL_INCLUDE_DIRS=$HOME/.conda/envs/mydartpyenv/include/fcl
```

Could be something easy to fix with the DARTFindfcl.cmake or Findfcl.cmake, or is it?

dgtim, 2020-11-09:
Versions:
```
(mydartpyenv) [build]$ conda list
# packages in environment at /home/degtyai1/.conda/envs/mydartpyenv:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
assimp                    5.0.1                hf01f27e_4    conda-forge
boost-cpp                 1.74.0               h9359b55_0    conda-forge
bzip2                     1.0.8                h516909a_3    conda-forge
c-ares                    1.16.1               h516909a_3    conda-forge
ca-certificates           2020.6.20            hecda079_0    conda-forge
certifi                   2020.6.20        py39h079e4ff_2    conda-forge
collada-dom               2.5.0                hdced9d9_3    conda-forge
console_bridge            1.0.1                hc9558a2_0    conda-forge
eigen3                    3.3.7                         0    omnia
expat                     2.2.9                he1b5a44_2    conda-forge
fcl                       0.6.1                h609abcd_2    conda-forge
ffmpeg                    4.3.1                h3215721_1    conda-forge
flann                     1.9.1             h44f99b7_1008    conda-forge
fontconfig                2.13.1            h7e3eb15_1002    conda-forge
freetype                  2.10.4               h7ca028e_0    conda-forge
gmp                       6.2.0                h58526e2_4    conda-forge
gnutls                    3.6.13               h79a8f9a_0    conda-forge
hdf5                      1.10.6          nompi_h54c07f9_1110    conda-forge
icu                       67.1                 he1b5a44_0    conda-forge
jpeg                      9d                   h36c2ea0_0    conda-forge
krb5                      1.17.1               hfafb76e_3    conda-forge
lame                      3.100             h14c3975_1001    conda-forge
ld_impl_linux-64          2.33.1               h53a641e_7  
libccd                    2.1                  he1b5a44_1    conda-forge
libcurl                   7.71.1               hcdd3856_8    conda-forge
libedit                   3.1.20191231         h14c3975_1  
libev                     4.33                 h516909a_1    conda-forge
libffi                    3.3                  he6710b0_2  
libgcc-ng                 9.1.0                hdf63c60_0  
libgfortran-ng            7.5.0               hae1eefd_17    conda-forge
libgfortran4              7.5.0               hae1eefd_17    conda-forge
libiconv                  1.16                 h516909a_0    conda-forge
libjpeg-turbo             2.0.5                h516909a_0    conda-forge
libnghttp2                1.41.0               h8cfc5f6_2    conda-forge
libpng                    1.6.37               h21135ba_2    conda-forge
libssh2                   1.9.0                hab1572f_5    conda-forge
libstdcxx-ng              9.1.0                hdf63c60_0  
libtiff                   4.1.0                h4f3a223_6    conda-forge
libuuid                   2.32.1            h14c3975_1000    conda-forge
libwebp-base              1.1.0                h36c2ea0_3    conda-forge
libxcb                    1.13              h14c3975_1002    conda-forge
libxml2                   2.9.10               h68273f3_2    conda-forge
lz4-c                     1.9.2                he1b5a44_3    conda-forge
ncurses                   6.2                  he6710b0_1  
nettle                    3.4.1             h1bed415_1002    conda-forge
octomap                   1.9.5                hc9558a2_1    conda-forge
openh264                  2.1.1                h8b12597_0    conda-forge
openscenegraph            3.6.5               hd27bf31_10    conda-forge
openssl                   1.1.1h               h516909a_0    conda-forge
pcre                      8.44                 he1b5a44_0    conda-forge
pip                       20.2.4           py39h06a4308_0  
pthread-stubs             0.4               h14c3975_1001    conda-forge
pybind11                  2.6.0            py39h3693e7b_0    conda-forge
pybind11-global           2.6.0                    pypi_0    pypi
python                    3.9.0                hcff3b4d_1  
python_abi                3.9                      1_cp39    conda-forge
readline                  8.0                  h7b6447c_0  
setuptools                50.3.1           py39h06a4308_1  
sqlite                    3.33.0               h62c20be_0  
tinyxml                   2.6.2                hc9558a2_1    conda-forge
tk                        8.6.10               hbc83047_0  
tzdata                    2020d                h14c3975_0  
urdfdom                   2.3.3                hc9558a2_0    conda-forge
urdfdom_headers           1.0.5                hc9558a2_2    conda-forge
wheel                     0.35.1                     py_0  
x264                      1!152.20180806       h14c3975_0    conda-forge
xorg-kbproto              1.0.7             h14c3975_1002    conda-forge
xorg-libx11               1.6.12               h516909a_0    conda-forge
xorg-libxau               1.0.9                h14c3975_0    conda-forge
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge
xorg-xproto               7.0.31            h14c3975_1007    conda-forge
xz                        5.2.5                h7b6447c_0  
zlib                      1.2.11               h7b6447c_3  
zstd                      1.4.5                h6597ccf_2    conda-forge

```

traversaro, 2020-11-09:
In https://github.com/conda-forge/dartsim-feedstock we build dart 6.9.5 with no problems, I wonder if there is some difference between 6.9.0 and 6.9.5 ?

traversaro, 2020-11-09:
It seems that fcl info is found via its .pc pkg-config file, so perhaps you need to install pkg-config in conda?

dgtim, 2020-11-09:
installing pkg-config did not help, still the same error message about missing fcl

jslee02, 2025-10-26:
Closing now that we have dartpy package in conda-forge: https://prefix.dev/channels/conda-forge/packages/dartpy

closed 2025-10-26 by jslee02