dartsim/dart #1535: [Err] [DARTPhysics.cc:760] The use of the ODE collision detector with DART is disabled because it causes conflicts with the version of ODE used in Gazebo.

trns1997, 2021-01-27:
I am using Gazebo 11.0.0 and have installed it from source. I wish to use ODE collision detection but I get he following error. I was wondering if someone could enlighten me with the following error. And if it is a problem with dart or just gazebo.

azeey, 2021-01-27:
This is an issue in Gazebo. See https://github.com/dartsim/dart/issues/1425#issuecomment-555672526.

trns1997, 2021-01-27:
Unfortunately the bitbucket link doesn't exist anymore. Do you by any change have any suggestions to resolve this conflict?

azeey, 2021-01-27:
The archived PR is at https://osrf-migration.github.io/gazebo-gh-pages/#!/osrf/gazebo/pull-requests/2956/page/1. I don't have a suggestion that seems tractable. Have you looked into using [ign-gazebo](https://github.com/ignitionrobotics/ign-gazebo)? It uses DART with ODE as the collision detector.

trns1997, 2021-01-28:
I will give it a try today and keep you posted, thanks

trns1997, 2021-01-29:
```
    <physics type="dart">
      <collision_detector>dart</collision_detector>
    </physics>
```

I opted for the following work around. Where I use darts collision detector, but replaced my collision mesh with a cuboid. Seems to do the job.